react-native-camera/react-native-camera #2717: RNCamera crashes in iOS

andrewterex20190813, 2020-02-28:
I'm developing a video recording app in React Native and I'm using 'react-native-camera' package. Of course, I used RNCamera and the thing is it's fine in android but crashes in iOS. When I open the screen where RNCamera is used, the app just crashes and turn off. I can't get information why this issue occurs. If anyone has experience with this problem, I beg your help. Thank you.

`"react": "16.9.0",
"react-native": "0.61.5",
"react-native-camera": "3.15.0"

<RNCamera
    ref={ref => {
        this.camera = ref;
    }}
    style={{ flex: 1 }}
    type={RNCamera.Constants.Type.front}
/>`

ponchoalv, 2020-03-09:
same here! but using Type.back

andrewterex20190813, 2020-03-14:
The problem is audio permission.

closed 2020-03-14 by andrewterex20190813

ZaikinaEvgeniya-2, 2021-09-22:
same issue here for ios

"react-native": "0.61.5",
"react-native-camera": "^4.2.0",

Info.plist
<key>NSCameraUsageDescription</key>
<string>text</string>
<key>NSLocationWhenInUseUsageDescription</key>
<string>text</string>
<key>NSMicrophoneUsageDescription</key>
<string>text</string>
<key>NSPhotoLibraryAddUsageDescription</key>
<string>text</string>
<key>NSPhotoLibraryUsageDescription</key>
<string>text</string>