react-native-camera/react-native-camera #2718: MirrorImage props not working in front camera on iOS devices

tayguohong, 2020-02-28:
Hi, I a currently implementing selfie feature by using front camera in both android and iOS platform. During my testing, i found out that mirrorImage props in options not working in iOS, which leads to image mirrored on preview. Can anyone have solution to fix this issue? Because on android, this props works fine and I need to standardise these behaviour. Thank you.  

"react-native": "0.60.4",
"react-native-camera": "^3.18.0"

SirKadogan, 2020-03-03:
Yup, I'm having this issue too. 

Also, there could be some clarification in the props. If you're using the front camera and set "mirrorImage = true", that actually means the image won't be mirrored. I assume that is because the default behavior for the front camera is to mirror, so by setting that prop, you're actually "unmirroring". MirrorImage works fine for the back camera, though.

tayguohong, 2020-03-04:
> Yup, I'm having this issue too.
>
> Also, there could be some clarification in the props. If you're using the front camera and set "mirrorImage = true", that actually means the image won't be mirrored. [...]

Yes i actually want to "unmirror" the image, just like the camera app in android phone. But since iOS platform ignore `MirrorImage` props, it will causing problem to standardise the selfie behaviour in both platform.

SirKadogan, 2020-03-06:
Did you come up with any solution? I tried looking for other libraries, like react-native-image-picker and react-native-image-crop-picker, but they open a camera screen and that doesn't fit my application.
I'm also looking for a way to mirror the image, but no results yet. Something like mirroring the image based on the data, idk.

tayguohong, 2020-03-10:
> Did you come up with any solution? I tried looking for other libraries, like react-native-image-picker and react-native-image-crop-picker, but they open a camera screen and that doesn't fit my [...]

No solution found yet. I am putting this issue aside as I have other modules to work with. But if I found any solution, I will let you know.

KristineTrona, 2020-04-08:
Also interested in this. I tried setting "mirrorImage=true", when using the selfie camera, but even then the image gets mirrored, I think this prop is indeed ignored on ios.

sibelius, 2020-04-08:
we gonna accept a pull request to fix this

tonybennett, 2020-04-16:
Adding `forceUpOrientation=true` fixed this for me. So `{mirrorImage: true, forceUpOrientation: true}` on `takePictureAsync`.

closed 2020-06-23 by stale[bot]

ashishprasad2163, 2021-09-18:
> Adding `forceUpOrientation=true` fixed this for me. So `takePictureAsync({mirrorImage: true, forceUpOrientation: true})`.

Worked for me 👍🏻