react-native-camera/react-native-camera #2728: Recording video on Android for about 30 min crashes app RN 0.61 Exception Type: Unknown (SIGSEGV) SENTRY

mauriciopf, 2020-03-05:
How to tell what crashes? It seems that is an error from JNI? So the JDK doesn't seem to have symbols in Sentry (Hermes engine is enable)

`Exception Type: Unknown (SIGSEGV)

Application Specific Information:
Segfault

Thread 0 Crashed:
0   libhermes.so                    0x7a882de570        <unknown>
1   libhermes.so                    0x7a883ec8c4        <unknown>
2   libhermes.so                    0x7a883ec7bc        <unknown>
3   libhermes.so                    0x7a883c3eac        <unknown>
4   libhermes.so                    0x7a882ee838        <unknown>
5   libhermes.so                    0x7a88341b60        <unknown>
6   libhermes.so                    0x7a8830ac6c        <unknown>
7   libhermes.so                    0x7a8834be60        <unknown>
8   libhermes.so                    0x7a882f6be4        <unknown>
9   libhermes.so                    0x7a8834f630        <unknown>
10  libhermes.so                    0x7a88346da0        <unknown>
11  libhermes.so                    0x7a88346450        <unknown>
12  libhermes.so                    0x7a8834fef4        <unknown>
13  libhermes.so                    0x7a88346da0        <unknown>
14  libhermes.so                    0x7a88346450        <unknown>
15  libhermes.so                    0x7a8834fef4        <unknown>
16  libhermes.so                    0x7a88346da0        <unknown>
17  libhermes.so                    0x7a88346450        <unknown>
18  libhermes.so                    0x7a8834fef4        <unknown>
19  libhermes.so                    0x7a88346da0        <unknown>
20  libhermes.so                    0x7a88326d74        <unknown>
21  libhermes.so                    0x7a88346a60        <unknown>
22  libhermes.so                    0x7a8834fed4        <unknown>
23  libhermes.so                    0x7a88346da0        <unknown>
24  libhermes.so                    0x7a88326d74        <unknown>
25  libhermes.so                    0x7a88346a60        <unknown>
26  libhermes.so                    0x7a8834fed4        <unknown>
27  libhermes.so                    0x7a88346da0        <unknown>
28  libhermes.so                    0x7a88346450        <unknown>
29  libhermes.so                    0x7a882e5ed0        <unknown>
30  libhermes-executor-release.so   0x7a88076770        <unknown>
31  libhermes-executor-release.so   0x7a880765cc        <unknown>
32  libhermes-executor-release.so   0x7a88071cc4        <unknown>
33  libhermes-executor-release.so   0x7a88074480        <unknown>
34  libreactnativejni.so            0x7a88268de8        <unknown>
35  libreactnativejni.so            0x7a8826a210        <unknown>
36  libreactnativejni.so            0x7a8822d524        <unknown>
37  libreactnativejni.so            0x7a8821cfcc        <unknown>
38  libreactnativejni.so            0x7a8821cf48        <unknown>
39  base.odex                       0x7a90a09780        <unknown>`

cristianoccazinsp, 2020-03-11:
Is that something you can reproduce consistently? Any chance there's a memory leak somewhere? Or perhaps something too big ends up written to memory or goes through the bridge?

mauriciopf, 2020-03-13:
@cristianoccazinsp yes it's an outOfMemory error, seems due to the **stringify** of another library **redux-persist** closing this issue thanks!

closed 2020-03-13 by mauriciopf