react-native-camera/react-native-camera #2731: Background threading is not stable for some LGE devices.

leeopop, 2020-03-08:
# Bug Report

**To Do First**

- [ ] Did you try latest release?
- [ ] Did you try master?
- [ ] Did you look for existing matching issues?

**Platforms**

Android 9
LG G7

**Versions**

I have tried every released versions,
and found that its working at 3.8.0 and fails for all later releases.

**Description/Current Behaviour**
Camera App freezes when trying to capture images fastly.
Especially when capturing while autofocus is working.
However, it turned out that autofocus is not the probem.

** Solution **
After dozens of tries, we found that its the problem of LGE that their proprietary code contains concurrency bug so that using background thread make their system unstable.
I found that bgthread is applied after 3.8.0.

I agree that using bgThread "theoretically" not to harm any android implementations, however there are some poorly written android kernels :(
My suggestion is to provide a fallback flag to `CameraView` constructor, just like `fallbackToOldApi` flag.
In the fallback mode, we disable bgthread and do all tasks synchronously.

If this suggestion is acceptable, I'm glad to prepare some PRs solving this issue.

leeopop, 2020-03-08:
I found some occasional crash reports that are hardly reproduced. I hope this may help those issues.

cristianoccazinsp, 2020-03-11:
Can you get some error logs or stack traces? Are you confident this is a concurrency issue? What concurrent execution is causing the issue exactly? As far as I know, there are various places where concurrency happens (async tasks and background threads for long blocking operations).  If we could reproduce the issue, perhaps a lock can be added in the right spot to prevent such crash.

leeopop, 2020-03-11:
@cristianoccazinsp Thank you for your response! 
We found some log messages saying that decoding JPEG is failed because the file has not completely written.
The logs were generated from proprietary code so that we couldn't find the exact location where it happened.
We were sure this is caused by concurrency as we were free from this bug with camera3.8.0.
The bug reproduced with any later version.

I'll try to provide reproduced error logs/materials and notify your again when prepared.

kingworm, 2020-03-11:
hi @cristianoccazinsp,
I'm a collaborator with @leeopop .

I used LGE device : LG V30.
I found this issues on react-native-camera v3.11.0 ~ 3.19.0.
I referenced logs in https://github.com/kingworm/react-native-camera-lg/issues/2
I'll try where this issue came from

thank you

cristianoccazinsp, 2020-03-11:
Is it a native crash, or does the app just freeze? 

> We found some log messages saying that decoding JPEG is failed because the file has not completely written.
Doesn't seem related to the background thread code updates. Remember that the background thread is used mostly for start/stop and settings updates, but the step to actually capture a picture was already happening in background (through an async task)

If you're using a crash reporting tool (e.g., bugsnag) you should be able to get at least the main Java exception that's causing the app to crash.

cristianoccazinsp, 2020-03-11:
Also, does this only happen when taking pictures rapidly? Can you share a small snippet of what your JS code looks like? Are you waiting for the camera to finish capturing before capturing a new picture?  Make sure to use `onPictureTaken` events to re-enable the camera button after it has been pressed.

cristianoccazinsp, 2020-03-11:
Lastly, if you have an easy way to reproduce the issue, you could try the following:

First, make sure the above is true.

Then, change this method locally and remove the `runnable` usage, just let the code flow on w/e thread it was called: https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/org/reactnative/camera/RNCameraView.java#L244  I believe this is the change with bg threads you're referring to.

Following, take a look at this class. This is the async task that handles pictures after capture. Note that this was NOT modified by any background thread update: https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/org/reactnative/camera/tasks/ResolveTakenPictureAsyncTask.java

Lastly, you may try removing this part as well to see if that's causing issues.  This was updated as part of speeding up the camera and perhaps has bugs with LG's implementations: https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L736

kingworm, 2020-03-12:
I recorded this issue with video. 
you can see the video here.
https://drive.google.com/open?id=1ySDYzWWQr4aqeEI1BH9VfEgYRBybZRuB

> Is it a native crash, or does the app just freeze?

I think the app just freeze.

> If you're using a crash reporting tool (e.g., bugsnag) you should be able to get at least the main Java exception that's causing the app to crash.

I used just adb, and I'll try crash reporting tool as well.

> Also, does this only happen when taking pictures rapidly? Can you share a small snippet of what your JS code looks like?

it happens when taking pictures so close to objects, or while moving the screen quickly ( I mean when the screen loses focus ).
and I tested in the example code provided by RNCamera Docs.
https://github.com/react-native-community/react-native-camera/issues/2731

> Are you waiting for the camera to finish capturing before capturing a new picture? Make sure to use `onPictureTaken` events to re-enable the camera button after it has been pressed.

I think this is not our case.

kingworm, 2020-03-12:
> Lastly, if you have an easy way to reproduce the issue, you could try the following:
>
> First, make sure the above is true.
>
> Then, change this method locally and remove the `runnable` usage, just let [...]

I'll follow your advice, and let you know. Thank you!

cristianoccazinsp, 2020-03-12:
@kingworm I've just looked at the video. It looks like you are not preventing the camera button from being "spammed" (it does not get disabled while the camera is capturing). You should definitely disable it before calling `takePictureAsync` and enable it back with the `onPictureTakenEvent` (if you don't want to wait for post processing) or just after `takePictureAsync` (if you want to wait for post processing).

My guess is that you observe this more frequently when capturing close objects because the camera has a hard time focusing, and hence, capturing takes longer. When you send another capture request while the camera is still not ready, you should normally get an error message, but looks like this is not the case for this device and it just acts odd. Please try the above and let me know.

cristianoccazinsp, 2020-03-12:
Lastly, it doesn't seem like a crash neither; looks like the preview is being stopped.  According to google (and the source code), the camera itself has to resume the preview after every capture. You are right that this might be a race condition in the code and the preview is not getting resumed properly when the camera is pushed hard like this (without waiting for capture events).

kingworm, 2020-03-16:
@cristianoccazinsp 
I've conducted some experiments, not changing npm src code.
1. as you said, I tried preventing the camera button from being spammed.
`
  const onPictureTaken = () => {
    onPicTaken(false);
  };

  const takePicture = async camera => {
    // not to being spammed.
    if (takingPic) {
      return;
    }
    onPicTaken(true);
    try {
      const options = {quality: 0.5, base64: true};
      const data = await camera.takePictureAsync(options);
      console.log(data.uri);
    } catch (err) {
      console.log(err.message);
    }
  };
`
it just locked the button, but the preview was still freezed.
it was clear that this happened from **takePictureAsync**.

2. I ignored the camera button get spammed, but I tried to unfreeze the preview screen when I captured close objects. 
`
  const takePicture = async camera => {
    // not to being spammed.
    // if (takingPic) {
    //   return;
    // }
    onPicTaken(true);
    try {
      const options = {quality: 0.5, base64: true};
      const data = await camera.takePictureAsync(options);
      console.log(data.uri);
    } catch (err) {
      console.log(err);
      camera.pausePreview();
      camera.resumePreview();
    }
  };
`
the console says, _[Error: Camera capture failed. Camera is already capturing.]_ but the screen unfreezed!!

I think it is just a temporary solutions(because I couldn't get image uri when catch phrase), but anyway it worked.
if you give an any help, let me know.
thank you!!

kingworm, 2020-03-16:
@leeopop

leeopop, 2020-03-17:
@cristianoccazinsp We first thought that there was no way to overcome this problem without changing RNCamera internals.
The workaround solution did work well and it seems we do not have to change the internals.

I'll close this issue for now and you may reopen it if we have more things to discuss.

closed 2020-03-17 by leeopop