react-native-camera/react-native-camera #2732: Android front facing camera landscape facial detection bug, possibly fixed?

maxdiebold, 2020-03-09:
# Bug Report

**To Do First**

- [ Check ] Did you try latest release?
- [ Check ] Did you try master?
- [ Check ] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: APK 23
- iOS:
- react-native-camera: Master
- react-native: 0.61.5
- react: 16.9.0
  react-navigation: 5.0.3

**Description/Current Behaviour**

Face detection on the front facing camera worked great, but didn't work in either landscape orientation. Face detection on the back camera worked fine in any orientation. The front facing detection looked like it was trying to render the bounding box on the opposite side, so I figured it had to do with a basic mirror calculation.

After some hunting, I found this in RNCameraViewHelper:

`public static int getCorrectCameraRotation(int rotation, int facing, int cameraOrientation) {
    if (facing == CameraView.FACING_FRONT) {
      // Tested the below line and there's no need to do the mirror calculation
      return (cameraOrientation + rotation) % 360;
    } else {
      final int landscapeFlip = rotationIsLandscape(rotation) ? 180 : 0;
      return (cameraOrientation - rotation + landscapeFlip) % 360;
    }
  }`

and thought that maybe it did indeed need the mirror calculation. Changed it to this:

`  public static int getCorrectCameraRotation(int rotation, int facing, int cameraOrientation) {
    if (facing == CameraView.FACING_FRONT) {
      // Tested the below line and there's no need to do the mirror calculation
      final int landscapeFlip = rotationIsLandscape(rotation) ? 180 : 0;
      return (cameraOrientation + rotation + landscapeFlip) % 360;
    } else {
      final int landscapeFlip = rotationIsLandscape(rotation) ? 180 : 0;
      return (cameraOrientation - rotation + landscapeFlip) % 360;
    }
  }`

and it works as expected now.

I tested this on a Samsung Galaxy S6. Has anyone else run into this problem? Would this change break behavior on any other devices?

**Expected Behaviour**

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

closed 2020-05-15 by stale[bot]

viktor-kurysh, 2020-07-23:
I have faced this issue too.
The up above solution helped me.
Thanks a lot!

xtealer, 2020-07-23:
Had this issue too, this worked like a charm!

vertvvv, 2020-10-29:
Not sure that this issue should be closed, cause bug is still here. Big thx for fix @maxdiebold !