react-native-camera/react-native-camera #2733: CI basic example - Duplicate module name: react-native

fabriziobertoglio1987, 2020-03-09:
# Bug Report
**Assigning to myself. I am writing myself a pull request to fix this bug.** 

Bug in Circle CI affecting pull requests [2724][1], [2723][2], [2581][3], [2569][4] *(I verified the Circle CI logs and they are all the same)*.

The gradle build both in CI and Local Environment has no issues, but `Task :app:bundleReleaseJsAndAssets` fails. 
The errors is caused from a name collision between the `examples/basic/node_modules/react-native/package.json` and `examples/advanced/advanced/node_modules/react-native/package.json`.

**To Do First**

- [X] Did you try latest release?
- [X] Did you try master?
- [X] Did you look for existing matching issues? [1767][6] seems related, but for iOS and it is closed.

**Platforms**

The bug is experienced on pull request for Android and Windows side (quoted above).

**Description/Current Behaviour**

The gradle build both in CI and Local Environment has no issues, but `Task :app:bundleReleaseJsAndAssets` fails. 
The errors is caused from a name collision between the `examples/basic/node_modules/react-native/package.json` and `examples/advanced/advanced/node_modules/react-native/package.json`.

```log
> Transform artifact okhttp-urlconnection.jar (com.squareup.okhttp3:okhttp-urlconnection:3.12.1) with JetifyTransform
ERROR: [TAG] Failed to resolve variable '${animal.sniffer.version}'
ERROR: [TAG] Failed to resolve variable '${project.groupId}'
ERROR: [TAG] Failed to resolve variable '${project.version}'
ERROR: [TAG] Failed to resolve variable '${project.groupId}'
ERROR: [TAG] Failed to resolve variable '${project.version}'
ERROR: [TAG] Failed to resolve variable '${project.groupId}'
ERROR: [TAG] Failed to resolve variable '${project.version}'
ERROR: [TAG] Failed to resolve variable '${project.groupId}'
ERROR: [TAG] Failed to resolve variable '${project.version}'

> Task :app:bundleReleaseJsAndAssets
warning: the transform cache was reset.
error jest-haste-map: Haste module naming collision:
  Duplicate module name: react-native
  Paths: /home/fabrizio/Documents/sourcecode/opensource/react-native-camera/examples/basic/node_modules/react-native/package.json collides with /home/fabrizio/Documents/sourcecode/opensource/react-native-camera/examples/advanced/advanced/node_modules/react-native/package.json

This error is caused by `hasteImpl` returning the same name for different files.. Run CLI with --verbose flag for more details.

> Task :app:bundleReleaseJsAndAssets FAILED
```
additionally I experimented the following bug running the [basic example][5], but I skipped as the advanced example works fine.

[1]: https://github.com/react-native-community/react-native-camera/pull/2724
[2]: https://github.com/react-native-community/react-native-camera/pull/2723
[3]: https://github.com/react-native-community/react-native-camera/pull/2581
[4]: https://github.com/react-native-community/react-native-camera/pull/2569
[5]: https://github.com/react-native-community/react-native-camera/issues/2547
[6]: https://github.com/react-native-community/react-native-camera/issues/1767

**Expected Behaviour**

CI builds without issues.

**Steps to Reproduce**

The bug can be reproduced by forking those pull request and running 

```bash
cd examples/basic/android
chmod +x ./gradlew && ./gradlew clean && ./gradlew build
```

sibelius, 2020-03-09:
try to use this https://github.com/react-native-community/react-native-camera/blob/master/examples/basic/rn-cli.config.js

fabriziobertoglio1987, 2020-03-09:
Thanks a lot. I am reading [cli/configuration][1] and [rnpm/rnpm][2] and I will keep working on this.

[1]: https://github.com/react-native-community/cli/blob/master/docs/configuration.md
[2]: https://github.com/rnpm/rnpm

MateusAndrade, 2020-03-09:
Hey @fabriziobertoglio1987 i just assigned this issue to you. Btw, thanks for your help.

I was looking at your PR and don't find anything specific that can break the CI build. Also #2723 looks fine when running the build at Circle. 

Is there anything that i can do to help you?

sibelius, 2020-03-09:
https://github.com/react-native-community/react-native-camera/blob/master/examples/basic/README.md#why-is-react-native-camera-not-listed-as-a-dependency-in-packagejson

closed 2020-03-10 by fabriziobertoglio1987