react-native-camera/react-native-camera #2734: Camera status is READY, but takePictureAsync throwing "Camera is not running"

SirCameron, 2020-03-09:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**
Android

**Versions**

- Android: Pie on emulator on mac os
- iOS: works fine here.
- react-native-camera: 3.19.0
- react-native: 0.61.5
- react: 16.9.0
  

**Description/Current Behaviour**
onStatusChange is responding that the camera is READY, but when attempting to take a picture with takePictureAsync, the error thrown is that the "camera is not running".

**Expected Behaviour**
For the 2 statuses to be in sync and the camera to take a picture.

**Steps to Reproduce**

Honestly, this is a big project so I can't run through that. It works perfectly on IOS.

SirCameron, 2020-03-09:
This is all ok on a real device. So not urgent.

cristianocca, 2020-03-11:
Shouldn't you be using `onCameraReady` instead?  I believe `onStatusChange` reports the permission status (and that the camera will eventually load), but it necessarily means that the camera has been fully initialized and ready to take pictures. 

Please let me know what you think. Perhaps the docs needs some updates in regard to these two events.

sonam1409, 2020-03-17:
I m facing below issue in android.
[image]

NooruddinLakhani, 2020-04-13:
pass function onCameraReady to RNCamera which trigger when camera ready for preview so you can start recording or take a picture then

renatobenks, 2020-06-12:
I'm closing this because what @NooruddinLakhani told above will make it work as you want @SirCameron. So, try to do that, and let us know, please!

closed 2020-06-12 by renatobenks