react-native-camera/react-native-camera #2736: upgrade basic example to ReactNative 0.61.5

fabriziobertoglio1987, 2020-03-10:
# Feature-Request

**Describe the Feature**

**I would like to assign this task to myself** and upgrade the Basic Example to `reactnative@0.61.5` for the following reasons:

1) The demo example is the main tool used to develop and test features for the react-native-camera library
2) The demo example is also used in Continuos Integration to run test on the device
3) Make the demo easier to install

[1]: https://github.com/react-native-community/react-native-camera/pull/2735
[2]: https://stackoverflow.com/questions/58051510/xcode-11-0-build-get-error-unknown-argument-type-attribute-in-method-r/58061998#58061998
[3]: https://github.com/react-native-community/react-native-camera/blob/master/docs/installation.md#ios---other-required-steps

**Possible Implementations**

1) Upgrade to react native 0.61.5 (solve [bug][2])
2) Add [permissions][3]
3) Update dependencies
4) Review the `metro.config.js` and `rn-cli.config.js` configurations

fabriziobertoglio1987, 2020-03-13:
Experienced on the master branch.
[image]

MateusAndrade, 2020-03-14:
> Experienced on the master branch and while working on pull [2724](https://github.com/react-native-community/react-native-camera/pull/2724)
> 
> [image]

Probably because of something between 3.9.0...master. Did you got any progress on that?

fabriziobertoglio1987, 2020-03-14:
I was working on upgrading the example app to rn0.61.5 but I'm postponing it. 

https://facebook.github.io/metro/docs/configuration/
https://callstack.com/blog/adding-an-example-app-to-your-react-native-library/

closed 2020-03-16 by fabriziobertoglio1987