react-native-camera/react-native-camera #2738: 3.19.1 breaks project

Juan-Castelli, 2020-03-12:
# Bug Report

**To Do First**

- [X] Did you try latest release?
- [X] Did you try master?
- [X] Did you look for existing matching issues?

**Platforms**

Doesn't matter because the bundle isn't even compiling so the Android version or platform is irrelevant, I think.

None.

**Versions**

- Android:  API 28
- iOS:
- react-native-camera: Works in 3.19.0, crashes completely in 3.19.1 (latest)
- react-native:0.61.5
- react:16.9.0
  
5.0.9

**Description/Current Behaviour**

FAILURE: Build failed with an exception.

What went wrong:
node_modules\react-native-camera\android\src\main\java\org\reactnative\camera\RNCameraViewHelper.java uses or overrides a deprecated API.
Execution failed for task ':react-native-camera:compileGeneralDebugJavaWithJavac'.
> Compilation failed; see the compiler error output for details.

**Expected Behaviour**

That it works in 3.19.1 the same as 3.19.0 I guess

**Steps to Reproduce**

Just try to npx react-native run-android and it breaks.

**Additionals**

I can provide any log or output necessary.

sibelius, 2020-03-12:
can you provide more info?

MateusAndrade, 2020-03-13:
@Juan-Castelli, @sibelius  we can look at the changelog from 3.9.0 to 3.9.1 here: https://github.com/react-native-community/react-native-camera/compare/v3.19.0...v3.19.1

Juan-Castelli, 2020-03-13:
> 
> 
> can you provide more info?

Do you need a specific log or output? @sibelius

cristianoccazinsp, 2020-03-14:
Seems to be compiling fine for me. I do see some warnings about some unsafe/unchecked operations, but it continues successfully.

puremana, 2020-03-19:
Same issue for me, v3.20.0 and v3.19.2 both didn't let me compile. Downgraded to 3.19.0 and it compiled fine. Android "react": "16.9.0", "react-native": "0.61.5", running on Nexus second gen. Sorry I didn't record the error messages

sibelius, 2020-03-19:
can you create a failing repro?

can you try to run examples?

MateusAndrade, 2020-03-19:
we can see for some change between https://github.com/react-native-community/react-native-camera/compare/v3.19.0...v3.20.0

cristianocca, 2020-03-19:
Any easy way to skip the `examples` folder? Lots of noise there. Any settings change that is now causing deprecated methods usage become errors?

https://github.com/react-native-community/react-native-camera/compare/v3.19.0...v3.19.2#diff-0991de55e949191b08ddeb020a0bf629R63

Could also look at similar issues from other repos. This definitely looks like a misconfig somewhere.

https://github.com/react-native-community/async-storage/issues/223
https://github.com/react-native-community/react-native-device-info/issues/325

Just a wild guess, but the people having issues most likely are missing the following somewhere:
```
lintOptions {
    abortOnError false
 }
```

closed 2020-05-26 by stale[bot]