react-native-camera/react-native-camera #2742: thoughts on splitting recording behavior / preview?

wolverineks, 2020-03-13:
# Question

**To Do First**
- [X] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [X] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [X] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**

any thoughts / is it possible to split the recording capabilities from the preview?
im thinking something like...

```typescript
const cameraSettings = { ... }
const Foo = () => {
  const [showPreview, setShowPreview] = React.useState(true)
  const { status, record, pause, stop } = useCamera(cameraSettings)

  return <View>
    <TouchableOpacity onPress={() => setShowpreview(showPreview => !showPreview)}>
      Toggle Preview
    </TouchableOpacity>
    {status === Camera.Status.isLoading && <ActivityIndicator />}
    {status !== Camera.Status.isRecording && <TouchableOpacity onPress={record}>Record</TouchableOpacity>}
    {status === Camera.Status.isRecording && <TouchableOpacity onPress={stop}>Stop</TouchableOpacity>}
    {status === Camera.Status.isRecording && <TouchableOpacity onPress={pause}>Pause</TouchableOpacity>}
    {status !== Camera.Status.isLoading && showPreview && <Camera.Preview style={styles.preview} />}
  </View>
}
```

**Tags**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

sibelius, 2020-03-13:
I would love `useCamera`, can you work on some pull requests to bring this new feature, without breaking the old behavior?

sibelius, 2020-03-13:
on hooks we have this 2 pull requests

https://github.com/react-native-community/react-native-camera/pull/2649
https://github.com/react-native-community/react-native-camera/pull/2646

not sure if this work can help you

wolverineks, 2020-03-13:
automated tests would really help a potentially substantial change like this
how are you verifying things now? manual q/a?

sibelius, 2020-03-13:
Manual QA for now

We could have native testing library and also detox setup

Do you wanna work on this first?

wolverineks, 2020-03-13:
it wouldnt hurt to start a separate discussion about setting up manual/automated/semi-automated testing

cristianoccazinsp, 2020-03-14:
As far as I know Android requires the preview layer to be running in order to take photos/videos.  Wouldn't it be a bit complicated to handle both? Looks like a good idea though!

wolverineks, 2020-03-14:
> As far as I know Android requires the preview layer to be running in order to take photos/videos. Wouldn't it be a bit complicated to handle both? Looks like a good idea though!

thats what im seeing as well, there are some posts about sending the video to a different type of android view, but that looks like it would be a change to react-native

the best i could figure is something like:
```typescript
 <RNCamera style={{ position: "absolute", opacity: 0, height: 100, width: 100 }} />
```
that could be turned into a prop like:
```typescript
 <RNCamera hidden />
```
or something like that

and if you really wanted something like:
```typescript
const { Camera, camera } = useCamera(...CAMERA_PROPS)

const takePicture = async () => {
  const uri = await camera.takePictureAsync()
  console.log({ uri })
}

return <View>
  {Camera}
</View>
```

but anyway you slice it, there has to be a component mounted in there somewhere
maybe if react-native had portals?

closed 2020-03-14 by wolverineks

cristianoccazinsp, 2020-03-14:
What were you trying to achieve anyways? Hidden recording? You can always hide the camera behind some view just like Facebook did a while ago (can't recall the bug, but it was hilliarious). However, I don't think that's really legal :D

wolverineks, 2020-03-14:
> What were you trying to achieve anyways? Hidden recording? You can always hide the camera behind some view just like Facebook did a while ago (can't recall the bug, but it was hilliarious). However, I don't think that's really legal :D

pretty much this, but dont worry, theres a disclaimer

fabriziobertoglio1987, 2020-04-01:
https://github.com/react-native-community/react-native-camera/issues/2178#issuecomment-579678651