react-native-camera/react-native-camera #2753: How to test with @testing-library/react-native?

prescindivel, 2020-03-16:
How to mock test with @testing-library/react-native? a

**To Do First**
- [ ] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [ ] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [ ] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**

**Tags**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

MateusAndrade, 2020-03-16:
You tried mocking mocking the implementation on RNCamera? I was facing a similar issue on a project, and i was able to do something like:

```js
jest.mock("path-to-library", () => mockedImpl)
```

You can try looking at: https://www.reactnative.guide/7-testing/7.4-mocking-rn-modules.html.

prescindivel, 2020-03-17:
yes, but i can't capture the takePictureAsync event

MateusAndrade, 2020-03-17:
> yes, but i can't capture the takePictureAsync event

You can provide any sample code? I have been struggling to test apps written in react-native using  some libraries. I would be happy to help you with that. 😄

danielholmes, 2020-03-26:
I've gotten the below working (with `expo-camera`). it works for a single test of recording a video - I don't know enough to say if it will work for multiple tests in a suite that use video recording.

`
let currentRecording: {
      resolve: (result: { uri: string }) => void;
      reject: (e: Error) => void;
    };
    jest.spyOn(Camera.prototype, 'recordAsync').mockImplementation(() => {
      return new Promise((resolve, reject) => {
        currentRecording = { resolve, reject };
      });
    });
    jest.spyOn(Camera.prototype, 'stopRecording').mockImplementation(() => {
      if (!currentRecording) {
        throw new Error('No recording');
      }
      currentRecording.resolve({ uri: '/local/video.mp4' });
    });
`

closed 2020-06-01 by stale[bot]