react-native-camera/react-native-camera #2754: Handle the fact that custom exif data is not included in the returned base64 property

IanPhilips, 2020-03-17:
# Feature-Request

**Describe the Feature**
If I pass an object containing GPS coords or other EXIF data to the writeExif optional parameter my file includes this exif data, though the data returned from `takePictureAsync` and accessed via the `base64` property does not. 

**Possible Implementations**
1. We could just note this in the RNCamera docs so people know what they're getting from the base64 property
2. We could write the exif into the base64 data so the `data.base64` returned after a `takePictureAsync` matches the base64 data loaded via `fs.readFile`

IanPhilips, 2020-03-18:
Regards to 2: I've been testing the library [piexifjs](https://github.com/hMatoba/piexifjs) for reading and editing base64 jpeg data and it works well. So far I can read the exif data out of the base64 representation of the jpeg, change tag values, (i.e. "GPSAltitude"), and rewrite the exif into the base64 data. This is a potential solution for images at least.

cristianoccazinsp, 2020-03-19:
Another option would be to use rn-fetch-blob file system to read the saved image as base64?

Also, can you tell me if this affects both iOS and Android? Looking at iOS' code, the data written to disk and memory seem to be the exact same chunk of data, so I'm not entirely sure why the extra exif data is lost there (https://github.com/react-native-community/react-native-camera/blob/master/ios/RN/RNCamera.m#L927)

For Android, the lack of implementation for base64 is definitely there.  Android makes it a bit harder since exif manipulation require the actual file on disk ( I think):
https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/org/reactnative/camera/tasks/ResolveTakenPictureAsyncTask.java#L214

For Android, perhaps a different approach can be taken to manipulate the exif data in memory, but that would require another library to do so (instead of Android's built in one)

IanPhilips, 2020-03-20:
True, we could use rn-fetch-blob. This would probably be the most foolproof way of doing it. In my experience editing/reading exif can be quite tricky. The exif I'm reading via piexifjs is actually only grabbing a subset of the total tags. 

For android the process of getting the exact same base64 after it's saved to the disk is like so:

   
     const data = await this.state.camera.takePictureAsync(options);
     const fileName = data.uri.slice(data.uri.lastIndexOf("/") + 1);
     const fullPath = "file:///storage/emulated/0/Pictures/" + fileName;
     RNFetchBlob.fs.readFile(fullPath, 'base64').then((data) => {
                data.base64 = data;
            })

I don't have an iOS device easily accessible atm so can't speak to that yet.

IanPhilips, 2020-04-16:
I can confirm this is not a problem on iOS, the file when transferred off the device and hashed gives the same exact hash as the base64 data.

closed 2020-06-23 by stale[bot]