react-native-camera/react-native-camera #2760: automated e2e testing

wolverineks, 2020-03-22:
# Feature-Request

**Describe the Feature**
automated testing with detox / jasmine

**Possible Implementations**
expo has 2 e2e apps, one for testing native changes (bare-expo) and one for testing javascript changes (test-suite)
https://github.com/expo/expo/tree/master/apps/
since this library requires native changes, bare-expo could be a viable model

heres an snack of a stripped down version of the test-suite 
https://snack.expo.io/@wolverineks/expo-tests

e2e at this point would be a single detox test:
1. wait until status === "done"
2. check if result === "passed"

of course, this would only cover non visual aspects of rncamera, but it could check that files are saved to disk with FileSystem, and that a \<Video \/\> component considers it a valid video file

here are all of the tests that expo has for expo-camera, (currently disabled, yikes!)
https://github.com/expo/expo/blob/master/apps/test-suite/tests/Camera.js

thoughts?

MateusAndrade, 2020-03-22:
This is really interesting, doing this we could deliver more stable release.

Btw, do you mean running a e2e app with detox l, ensuring that the changes on any PR don't break the project?

wolverineks, 2020-03-22:
im not too familiar with what _could_ be done with something like this. ive only ever used husky for automated testing locally, and manual deployments.

wolverineks, 2020-03-25:
[image]
[image]

wolverineks, 2020-03-25:
a bit of progress

fabriziobertoglio1987, 2020-03-26:
the problem with Detox e2e tests is that currently react-native-camera has only 1 file src/Camera.js which is javascript code, while most of the library is Java and ObjectiveC code. So using Detox would take a lot of memory from our CircleCI test suite but not making many assertion on our codebase.

So e2e would be very useful, but we still have a lot of Java and ObjectiveC code that needs to be mantained and tested... 

Hopefully I will start writing tests soon for the related Java functionalities I am mantaining...I would be happy that to see e2e tests too, but we need to understand how much will take our CircleCI to run them.

MateusAndrade, 2020-03-27:
> I am in favour of testing
> 
> I would like to start writing Unit Test with jUnit and mockito once I am done with #2766
> 
> I believe Testing is essential, as the library will always grow in complexity with the Machine Learning and AI features

Would you mind helping me on that? I have been working with JS for a long time, but i would love to learn more about tests on both Android and iOS. 🚀 

Also, this would be a great improvement for this project, since this can became a example to other projects.

sibelius, 2020-03-27:
@wolverineks  are you gonna bring this test case to react-native-camera?

we'd love to have them here

wolverineks, 2020-03-27:
I was thinking if it works well here, then add it to react-native-community/bob. But not all libraries need this type of testing. So, I'm still looking into it.

fabriziobertoglio1987, 2020-03-27:
@MateusAndrade I would use Junit and Mockito Unit Test covering the codebase that I am actively mantaining. If you need any help, we can code review each other.

I would keep things as simple as possible, for example as I am discussing in https://github.com/react-native-community/react-native-camera/pull/2766 

Writing and linking to the Android/IOS api we are using in the documentation so that the React Native developers can research information, writing additional wiki and documentation with examples on how to implement the features