react-native-camera/react-native-camera #2762: Unknown alert when rendering a screen containing a video with audio

lfoliveir4, 2020-03-23:
I have a screen in my application that when it is rendered, I use the react native camera to record the videos and on the same screen a video is already being recorded with audio. The library triggers an unknown alert.

How to resolve this alert?

[image]

cristianoccazinsp, 2020-03-24:
Try with `captureAudio=false`

lfoliveir4, 2020-03-25:
> Try with `captureAudio=false`

@cristianoccazinsp this is as props?

cristianoccazinsp, 2020-03-25:
Yes. Sorry about my previous short response. If I understood right, you have two video recordings? The error you have provided is due to issues in gaining access to the microphone, which is most likely because it is being used by another process.

lfoliveir4, 2020-03-25:
@cristianoccazinsp After passing this prop, the warning ceased to exist. But my question is:

When I record a video, using the camera of the cell phone will the audio in the environment be captured?
Or does this prop make this possibility unfeasible?

My use case is as follows: I open a screen that contains a video and on the same screen there is a button where I can record a video too and this is where this warning occurs.

closed 2020-05-31 by stale[bot]