react-native-camera/react-native-camera #2763: `onGoogleVisionBarcodesDetected()` returning wrong bounds (X and Y coordinates)

helciofranco, 2020-03-24:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**
Android

**Versions**

- Android: 9.0
- react-native-camera: 3.21.0
- react-native: 0.61.5
- react: 16.9.0
- react-navigation: 4.0.10

**Description/Current Behaviour**
`onGoogleVisionBarcodesDetected()` method is giving wrong barcodes bounds, both origin X and Y coordinates are wrong.

Try to test putting a barcode in front of your camera in all possible positions.
Test the edge top, edge left, edge bottom, middle...

You'll see the barcode in the middle (or bottom) of screen will not work. It returns a Y and X bigger than really it is.

**Expected Behaviour**
Should return the correct X and Y position of barcode, just as it is when using 'general' on _missingStrategy_.
Try `missingDimensionStrategy 'react-native-camera', 'general' ` to check all bounds are correct.

**Steps to Reproduce**
You can just start a project from scratch with RN-camera and use following _missingDimensionStrategy_ on your `android/app/buid.gradle`:
`missingDimensionStrategy 'react-native-camera', 'mlkit' `

**Additionals**
- In previous versions, it never did ocurred.
- Of course, app is 100% configured with google services, firebase, and all things related.
- I don't use Expo. 
- It only happens if I use _"mlkit"_ instead of _"general"_ on `missingDimensionStrategy`.
- iOS is perfect, not affected on this issue.

fabriziobertoglio1987, 2020-03-26:
https://github.com/react-native-community/react-native-camera/pull/2719#discussion_r391133195

closed 2020-06-02 by stale[bot]

ahmad-crossplatform, 2020-09-16:
Any news regarding this issue ? I still have it here

chiubaka, 2021-11-18:
I've got a fix for this that I can PR. The current calculations to map the image coordinates to preview coordinates are just plain wrong.

We're actually working between 3 different coordinate systems:
1. The image / frame captured from the camera
2. The Native Android View representing the camera preview
3. The React Native JS View representing the camera preview

Typically the resolution of images captured by the camera is much larger than the space made available by app developers for the preview. For example, my phone's camera takes 2048x1536 (4:3 aspect ratio) pictures.

Logging on device shows that, in my particular case, the RNCameraView occupies 1080x1797px, which is _not_ a 4:3 aspect ratio.

To make the actual preview image fit into the preview view, the image is resized using an "aspect fill" method. This means that the original image is scaled until one of its dimensions matches the dimensions of the preview view, then it's centered, and then any overflow is cropped away.

Once coordinates have been transformed to reverse the "aspect fill", they have to be transformed once more to account for scaling of pixels based on pixel density, as React Native seems to report view dimensions that differ from native Android dimensions by a factor of the screen pixel density.