react-native-camera/react-native-camera #2767: Access denied finding property "vendor.camera.aux.packagelist"

Hitesh2297, 2020-03-26:
Not able to open camera or see camera preview
Instead of showing white screen

lauwen, 2020-06-05:
Hi,I want to know how do you solve this problem

Hitesh2297, 2020-06-10:
i have not resolved this problem, i have removed this lib and feature as
well in application.

macortesn, 2020-06-26:
I have the same problem. Has anyone been able to fix it?

S0c5, 2020-06-26:
I'm having this, does anyone work on fix this?

JorgeVV, 2020-06-26:
Same issue but black screen instead.

niquefaDiego, 2020-06-29:
Same is happening to me

LouiseBC, 2020-07-08:
In case this helps anyone: I had the same problem on a Nokia 7.2 (other Android devices were fine) and it turned out to be a device bug. Restarting it fixed the problem

maheshappzoro, 2020-08-09:
> Not able to open camera or see camera preview
> Instead of showing white screen
> 
> In android studio log cat I found below error
> Access denied finding property "vendor.camera.aux.packagelist"
> 
> Access denied finding property "vendor.camera.hal1.packagelist"

Its fixed by this 👍 
android:hardwareAccelerated="true"

closed 2021-01-06 by stale[bot]