react-native-camera/react-native-camera #2773: Crash when pass codec and bitrate

sandeepsingh91, 2020-03-30:
# Bug Report

This is my code for record Options.

 const options = {};
        const dim = Dimensions.get('window');

        if (Platform.OS === 'ios') {
            options['videoBitrate'] = 0.8 * 1000 * 1000
            options['quality'] = RNCamera.Constants.VideoQuality['4:3']

            const majorVersionIOS = parseInt(Platform.Version, 10)

            if (majorVersionIOS >= 11) {
                if (dim.width >= 800) {
                    options['codec'] = RNCamera.Constants.VideoCodec['HVEC']
                } else {
                    options['codec'] = RNCamera.Constants.VideoCodec['H264']
                }
            } else {
                options['codec'] = RNCamera.Constants.VideoCodec['H264']
            }

        } else {
            options['quality'] = RNCamera.Constants.VideoQuality["720p"]
            options['videoBitrate'] = 3 * 1000 * 1000
        }

It work fine in Android but when I run this in iOS app got crashed.  Here are the logs from Xcode

*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[AVCaptureMovieFileOutput setOutputSettings:forConnection:] The only supported key is AVVideoCodecKey - the following keys are illegal: {(
    AVVideoCompressionPropertiesKey
)}'
*** First throw call stack:
(0x1c0f3b3d 0x1b37b067 0x22a1d3bd 0x2f2acd 0xb75467 0xb8166f 0xb78a39 0xb8376b 0xb83471 0x1b97687d 0x1b97645c)
libc++abi.dylib: terminating with uncaught exception of type NSException

cristianoccazinsp, 2020-03-31:
Can you try without `videoBitrate`? I don't understand why the error happens, but it seems like your device/OS version does not support setting the bitrate.

I would suggest a PR that changes the following with a try/catch and fallsback to only setting the codec.
https://github.com/react-native-community/react-native-camera/blob/master/ios/RN/RNCamera.m#L1147

sandeepsingh91, 2020-04-06:
@cristianoccazinsp when I tried without video bitrate it works fine. But I've to record video using custom bitrate.

closed 2020-06-12 by stale[bot]

renatobenks, 2020-06-12:
@sandeepsingh91 are you still facing it?

fonov, 2020-11-30:
I also have this issue