react-native-camera/react-native-camera #2777: PDF_417 barcode scan and issue with data and Canadian drivers licenses

bradwall17, 2020-04-07:
# Bug Report

**To Do First**

- [ ] Did you try latest release?
   yes
- [ ] Did you try master?
  Not yet
- [ ] Did you look for existing matching issues?
  yes

**Platforms**

Android 7.0 and up

iOS 10 and up

**Versions**

- Android:
- iOS:
- react-native-camera: 3.22.0
- react-native:  0.62.1
- react:  16.9.0
  

**Description/Current Behaviour**

When I try to scan a drivers license (PDF_417 barcode) for some Canadian drivers licenses, it has issues with the data.  When I scan and get data back, the data cuts off is some situations.
All the codes that I scan are ok, unless it has a weird character in it.  For example, if it looks like this (I took out some characters to protect data sensitivity):
@
ANSI 99<somenumbers>0001DLÖDLDC<other characters>
<the rest of the data is fine>
So, when I get the data back from the scan, it just gives me back:
@
ANSI 99<somenumbers>0001DL

The rest of the data is gone.  It is like the:  Ö is messing things up.  I don't see this in any US drivers licenses and they all come across fine.  The funny thing is that when I go into debug mode and go line by line in my code, when the data comes from the scanner, it comes across fine and all the data is there.  It is only when I am not in debug mode that it gets messed up.  I have even had my code do this:
<RNCamera ...
...
onBarCodeRead={(scanResult) => this.onBarCodeRead(scanResult)}
   ...
/> 

this.onBarCodeRead(scanResult) {
     console.log('data: ' + scanResult.data);
}

It just cuts off right before that strange character.

**Expected Behaviour**

**Steps to Reproduce**

We don't use Expo

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

jonzl, 2020-04-09:
Seems to be same issue as described in https://github.com/react-native-community/react-native-camera/issues/2761

JayeshPatil-Dev, 2020-09-23:
Driving license barcode scanning not working with react native camera.
PDF_417 barcode scan. USA driving license scan

closed 2020-10-01 by stale[bot]