react-native-camera/react-native-camera #2778: Android - takePictureAsync returns incomplete uri

baiross, 2020-04-08:
# Bug Report

**To Do First**

- [Yes ] Did you try latest release?
- [ Yes] Did you try master?
- [ Yes] Did you look for existing matching issues?

**Platforms**

**Versions**

- Android: 10
- iOS:
- react-native-camera: 3.22.0
- react-native: 0.16.4
- react:16.9.0
  

**Description/Current Behaviour**

At first capture, I can preview the captured image using FastImage component, but if I will try to capture again, the first captured image will not be deleted and when I log the response, the data is like this 'uri: "file:///data/user/0/com.aientry/cache/Camera"'
I don't know why there is no generated name after Camera. 
I can't save image too using CameraRoll :( 
**Expected Behaviour**

Can preview the captured images directly.

**Steps to Reproduce**

1. Open Camera
2. 1st Capture
3. The image can be seen using FastImage
4. 2nd Capture, the first captured image appears instead of the new one.

**Additionals**

Code: 
[image]

I created a new component (Camera)
[image]

[image]

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

cristianoccazinsp, 2020-04-08:
Haven't seen this happening.  Can you be more specific?  You take a picture, the first one comes with a good URI, but the second time you capture the file name is missing from the URI? Have you tried other devices?

Also, did you have a typo when writing RN's version? `react-native: 0.16.4` ?

baiross, 2020-04-08:
> Haven't seen this happening. Can you be more specific? You take a picture, the first one comes with a good URI, but the second time you capture the file name is missing from the URI? Have you tried other devices?
> 
> Also, did you have a typo when writing RN's version? `react-native: 0.16.4` ?

sorry I updated it. 61.4.

First capture = preview will display
Second capture = still first preview will display and so on

cristianoccazinsp, 2020-04-08:
What test device are you using?  Can you try without any delete/preview code? Just log the URI after each capture? Does the same happen?

cristianoccazinsp, 2020-04-08:
I wonder if this is related: https://github.com/react-native-community/react-native-camera/pull/2769

More specifically, https://github.com/react-native-community/react-native-camera/pull/2769/files#diff-e57ecb77f750149a9807397c9a79c9dbL183

@AsminBudha thoughts?

baiross, 2020-04-08:
> What test device are you using? Can you try without any delete/preview code? Just log the URI after each capture? Does the same happen?

I am using S10 and emulator with Android 9 version. 
I tried logging it and the weird thing is no file name after /Camera in uri.

cristianoccazinsp, 2020-04-08:
Can you try downgrading to 3.21.0 and see if the issue goes away?

baiross, 2020-04-08:
> Can you try downgrading to 3.21.0 and see if the issue goes away?

Hello, it's already working on 3.21.0. Thank You so much

cristianoccazinsp, 2020-04-08:
So 3.22.0 (https://github.com/react-native-community/react-native-camera/pull/2769) introduced an issue here. I can't still reproduce it myself though.

baiross, 2020-04-08:
did you try to delete the node_modules and package-lock.json ??

cristianoccazinsp, 2020-04-08:
Yes, and at first glance, although that PR has a bug, I can't see how the changes would end up giving you an empty file path. Might be a device specific issue.

cristianoccazinsp, 2020-04-08:
@baiross  can you try again with 3.22.0? This time, make sure you have `  <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />` in your android manifest file.

According to the following, `isDirectory()` might return false under some circumstances, which explains your problem.  Are you deleting the cache directory by any chance?

https://stackoverflow.com/questions/6987727/file-isdirectory-always-false-in-android-with-the-path

prescindivel, 2020-04-08:
I upgraded to version 3.22.1 and solved it for me

device: xiaomi mi a2
os: android 10

closed 2020-04-08 by sibelius

dnero, 2020-04-19:
> > Can you try downgrading to 3.21.0 and see if the issue goes away?
> 
> Hello, it's already working in 3.21.0. Thank You so much

@baiross was this the only thing you did to get it to work? I'm on:
Android 10 / Pixel 3 XL
"react": "16.9.0",
"react-native": "0.61.5",
"react-native-camera": "3.22.0",

and I've tried quite a few combinations of downgrades to no avail. I'm still seeing "file:///data/user/0/(reverse domain)/cache/Camera" with no image name as you were.

cristianoccazinsp, 2020-04-19:
I believe the fix was merged on v3.22.1 ?

Make sure you you are not running a cached version of the library also.

dnero, 2020-04-20:
Thanks @cristianoccazinsp - v3.22.1 did the trick.