react-native-camera/react-native-camera #2779: Unable to scan GS1 barcodes in iOS

pan23, 2020-04-08:
# Bug Report

Testing OS: 13.3
The Latest used  RNCamera version: 3.22.0

Unable to scan GS1 barcodes in iOS but working fine in Android with the same version.  I am not getting any errors as well. The camera keeps trying to scan the barcode for a long time but nothing happens.

sibelius, 2020-04-08:
did it worked before?

pan23, 2020-04-08:
No, for iOS it's not worked before as well

sibelius, 2020-04-08:
it worked before for android?

can you share the barcode?

pan23, 2020-04-09:
Yes, it's working fine on Android. Please check the attached image of the barcode. This image is scanning properly in android but not in iOS.
[image]

pan23, 2020-04-13:
Hi Sibelius, Have you checked the attached sample barcode? We are unable to scan these types of the barcode in iOS. Can you please suggest any solution for this?

dhawalsonawane, 2020-04-14:
@pan23 I am facing the same issue. Did you get any work around yet?
@sibelius could you provide any help on this?

blackgibbon, 2020-05-26:
what is the status on this?

closed 2020-08-02 by stale[bot]

sandy105, 2021-01-15:
@pan23 I am facing the same issue. Did you get any work around yet?
@sibelius could you provide any help on this?

zjw199011, 2021-02-03:
Does anyone figure out how to fix this? or maybe a work around?

ShahidNawazEteam, 2021-02-23:
Facing the same issue, working perfectly on Android but nothing happening on iOS. QR codes and some other barcode types are working fine but GS1 barcode types are not getting scanned.

Does anyone have a fix for this issue?

pan23, 2021-02-26:
I didn't found any solution with this library. I switched to another library which supports GS1 barcode for iOS.

ShahidNawazEteam, 2021-02-26:
@pan23 

Can please tell the name of the library the supports GS1 barcodes for IOS.

Thanks

sandy105, 2021-03-11:
@pan23  Could u tell me the name of the library that supports GS1 barcodes for IOS？ I used  zxingObject for iOS， but still have a issue on iPhone 12 pro.

pan23, 2021-03-11:
@sandy105 and @ShahidNawazEteam I used @yyyyu/react-native-barcode this library. Its working properly on iPhone 12 Pro with RN version 0.63.3. This library doesn't have any updates but still its working properly.

eteamIH, 2021-03-11:
Hi @pan23

Thanks for sharing this, can you kindly share the `formats` configuration that worked for you to scan on iPhone 12 Pro? The template is mentioned below.

```javascript
state = {
    visible: true,
    scanEnable: true,
    scannerSize: 300,
    decoder: RNLBarCode.Decoder.Auto,
    formats: RNLBarCode.Type.Common.QR_CODE,
    torch: RNLBarCode.TorchMode.Off
  };
```
Thanks!

eteamIH, 2021-03-11:
Also @pan23 , it would be great if you could kindly share your linking process for react-native-barcode library.

I'm also using RN 0.63 and it's not scanning GS1 in default settings. But when I say
```javascript
decoder: RNLBarCode.Decoder.ZBar
```

it gives me an error "Failed to invoke, this device doesn't support this decoder".

Any help in this regard will be highly appreciated.

Thank you!

sandy105, 2021-03-17:
@eteamIH  Zbar does not support the GS1 format of barcode， you can check it out on http://zbar.sourceforge.net