react-native-camera/react-native-camera #2780: Android Camera1 race condition crash

cristianoccazinsp, 2020-04-08:
# Bug Report

**To Do First**

- [X] Did you try latest release?
- [X] Did you try master?
- [X] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: 9
- react-native-camera: 3.21.0
- react-native: 0.61.5
- react: 16.13.1
  

**Description/Current Behaviour**

Looks like if the camera is unmounted/stopped at the same time a capture event happens, a race condition can trigger the following crash:
```
java.lang.RuntimeException Camera is being used after Camera.release() was called 
    Camera.java:-2 android.hardware.Camera.startPreview
    Camera1.java:764 com.google.android.cameraview.Camera1$7.onPictureTaken
    Camera.java:1207 android.hardware.Camera$EventHandler.handleMessage
    Handler.java:106 android.os.Handler.dispatchMessage
    Looper.java:214 android.os.Looper.loop
    HandlerThread.java:65 android.os.HandlerThread.run
```

**Expected Behaviour**
No crash.

**Steps to Reproduce**

Could not directly reproduce the issue myself since it comes from a race condition, but if we review the code, the race condition is quite obvious.

The following code will still execute after camera unmount: https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L763

Looking at the above crash, `startPreview` was called after the camera was stopped (https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L262)

Possible fix: Add a synchronized + null checks on the take picture callback. Our camera object should be null after being stopped as long was use the synchronized block.

@sibelius  what do you think about this? I will be working on the crash fix shortly since we've been receiving a few crashes from this in the past week.

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

sibelius, 2020-04-08:
is there a way to automate this kind of tests?

I think we need more synchorization to solve this race conditions

cristianoccazinsp, 2020-04-08:
I really don't think there's an easy way to automate this. Do emulators even run code in parallel / simulate multi-core architectures?

sibelius, 2020-04-08:
maybe detox

cristianoccazinsp, 2020-04-08:
Isn't detox for iOS simulator only? This kind of bug needs a real device/camera, so it's quite tricky to test.

As an update, I've been able to reproduce the race condition quite easily on a Pixel 2.  Pressing the app drawer button while the camera is still capturing will cause this crash.

sibelius, 2020-04-08:
Work both on android and iOS and simulator and real devices

closed 2020-06-14 by stale[bot]