react-native-camera/react-native-camera #2786: onBarCodeRead / onGoogleVisionBarcodesDetected should also send raw photo bytes

ceefour, 2020-04-11:
# Feature-Request

**Describe the Feature**

Currently `onBarCodeRead` only gives metadata, but not the image (bytes/base64) itself.
Calling `takePictureAsync()` does not give the real photo that was detected.

**Possible Implementations**

1. Save the image(s) in a bounded queue, make an optional prop to set format, i.e. `detectedImageFormat` = `none` (default) | `binary` | `base64`.
2. If barcode is detected, provide the image to the callback handler.

**Related Issues**

#2763, #2018.

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this feature to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

ceefour, 2020-04-11:
I think I know how to fix this, I'll post a PR for this.

ceefour, 2020-04-12:
Relevant symbols:

## Barcode Scanner

## Google Cloud Vision

* `org.reactnative.camera.tasks.BarcodeDetectorAsyncTask#serializeEventData`, which should prepare the `base64` and will call:
* `org.reactnative.camera.RNCameraView#onBarcodesDetected`, which will call:
* `org.reactnative.camera.RNCameraViewHelper#emitBarcodesDetectedEvent`, which will call:
* `org.reactnative.camera.events.BarcodesDetectedEvent#obtain`

ceefour, 2020-04-12:
Currently working on this on https://github.com/ceefour/react-native-camera/tree/hendy

cristianoccazinsp, 2020-04-23:
I like the idea.  What about the image's quality? Base64 seems like a lot of overhead for the bridge/json if the quality is high (similar to the one from takePictureAsync). Seems like you should pass in the same options as takePictureAsync.

ceefour, 2020-04-23:
@cristianoccazinsp Work is already done for both both BarCode and ML Kit Barcodes in this branch:
https://github.com/ceefour/react-native-camera/tree/hendy

After tidying up the commit I'll make a PR here.

musyilmaz, 2020-05-07:
I am currently trying to make your branch work though the event object in onBarcodeRead still no returning any base64, or am I doing something wrong? @ceefour

ceefour, 2020-05-07:
@musyilmaz make sure to set `detectedImageInEvent=true`

Thanks for reminding, I should make the PR tonight or tomorrow.

It's really encouraging to see this functionality needed by other users. :)

musyilmaz, 2020-05-07:
Actually this didnt work also. Checked your commit and it seems it only works in android? maybe after your pr, I might work on the ios section in where I was doing the tests. 

Also one another issue I encountered, when I used your branch as a package, it wasn't showing up any of the changes :)

I will wait for your pr to test it and maybe utilize. 

Currently, thanks to the async logic on takePicture and onBarcodeRead, when device reads barcode, a base64 picture from the camera can be taken like 1 second later.

ceefour, 2020-05-07:
@musyilmaz you are correct I only implemented on Android since that's my requirement at that time. Also I don't have access to iOS device.

> Currently, thanks to the async logic on takePicture and onBarcodeRead, when device reads barcode, a base64 picture from the camera can be taken like 1 second later.

I did try this before, and some people may be happy with this. But in my requirement it did not work, as I want to crop the exact barcode, and if there's a delay even just 1 frame the crop will be off.

manishkumar03, 2020-06-07:
I am soon going to create a PR for this for the iOS part. My plan is to return a `uri` of the image as part of the returned barcodes object itself i.e `barcode.uri`. It will be on by default as it adds almost no overhead.

manishkumar03, 2020-06-07:
Implemented this on iOS in [this PR](https://github.com/react-native-community/react-native-camera/pull/2864).

ceefour, 2020-07-23:
@musyilmaz please check #2923

closed 2020-07-23 by sibelius

n1ru4l, 2020-07-23:
:tada: This issue has been resolved in version 3.34.0 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v3.34.0)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

commit 0df72c31 (2020-07-23): feat(android): onBarCodeRead should also send raw photo bytes, fixed #2786 (#2923)

* feat: onBarCodeRead should also send raw photo bytes, fixed #2786

* docs: documentation to explain feature #2786