react-native-camera/react-native-camera #2787: Can't record, take pictures and create media file on Android 9 Xioami

lucastonon, 2020-04-11:
# Bug Report

**To Do First**

- [ Y ] Did you try latest release?
- [ N ] Did you try master?
- [ Y ] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: 9
- Device: Xiaomi Redmi 8A
- react-native-camera: 3.22.1
- react-native: 0.61.0
- react: 16.9.17

My manifest permissions:
```
<uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.ACCESS_NETWORK_STATE" />
    <uses-permission android:name="android.permission.SYSTEM_ALERT_WINDOW"/>
    <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.DOWNLOAD_WITHOUT_NOTIFICATION" />
    <uses-permission android:name="android.permission.CAMERA" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
    <uses-permission android:name="com.google.android.c2dm.permission.RECEIVE" />
    <uses-permission android:name="android.permission.RECORD_AUDIO"/>

    <uses-permission android:name="android.permission.ACCESS_WIFI_STATE" />

    <uses-feature android:name="android.hardware.camera" />
    <uses-feature android:name="android.hardware.camera.front" android:required="true" />
    <uses-feature android:name="android.hardware.camera.autofocus" />
```

  

**Description/Current Behaviour**

I am only having problem with a Xiomi device using Android 9,
the camera preview opens fine but it won't record nor take pictures.
Everything is working fine on Samsung Android 8 and on iOS

When I open the camera I see this:
```
E/libc: Access denied finding property "vendor.camera.aux.packagelist"
W/RNCamera-Handle: type=1400 audit(0.0:32297): avc: denied { read } for name="u:object_r:vendor_camera_prop:s0" dev="tmpfs" ino=12603 scontext=u:r:untrusted_app:s0:c176,c256,c512,c768 tcontext=u:object_r:vendor_camera_prop:s0 tclass=file permissive=0
E/libc: Access denied finding property "vendor.camera.aux.packagelist1"
E/libc: Access denied finding property "vendor.camera.aux.packagelist2"
E/libc: Access denied finding property "vendor.camera.aux.packagelist"
E/libc: Access denied finding property "vendor.camera.aux.packagelist1"
E/libc: Access denied finding property "vendor.camera.aux.packagelist2"
E/libc: Access denied finding property "vendor.camera.aux.packagelist"
E/libc: Access denied finding property "vendor.camera.aux.packagelist1"
E/libc: Access denied finding property "vendor.camera.aux.packagelist2"
E/libc: Access denied finding property "vendor.camera.hal1.packagelist"
W/ExifInterface: Stop reading file since a wrong offset may cause an infinite loop: 0
```

Then when I try to take a picture I get:

```
W/System.err: java.io.IOException: Couldn't create directory '/data/user/0/com.myapp/cache/Camera'
W/System.err:     at org.reactnative.camera.utils.RNFileUtils.ensureDirExists(RNFileUtils.java:19)
W/System.err:     at org.reactnative.camera.utils.RNFileUtils.getOutputFilePath(RNFileUtils.java:25)
W/System.err:     at org.reactnative.camera.tasks.ResolveTakenPictureAsyncTask.getImagePath(ResolveTakenPictureAsyncTask.java:320)
W/System.err:     at org.reactnative.camera.tasks.ResolveTakenPictureAsyncTask.doInBackground(ResolveTakenPictureAsyncTask.java:183)
W/System.err:     at org.reactnative.camera.tasks.ResolveTakenPictureAsyncTask.doInBackground(ResolveTakenPictureAsyncTask.java:27)
W/System.err:     at android.os.AsyncTask$2.call(AsyncTask.java:333)
W/System.err:     at java.util.concurrent.FutureTask.run(FutureTask.java:266)
W/System.err:     at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
W/System.err:     at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
W/System.err:     at java.lang.Thread.run(Thread.java:764)
W/ReactNativeJS: Possible Unhandled Promise Rejection (id: 1):
    Error: An unknown I/O exception has occurred.
```

And if I try to record I get:
`W/ReactNativeJS: 'record error', [Error: Starting video recording failed - could not create video file.]`

saimonventura, 2020-05-22:
Infinity Screen Bug?

closed 2020-07-29 by stale[bot]

tpharaoh, 2020-10-30:
I have almost identical issue with android 10. My Nokia was working, then it got a security patch. Now the same errors

E/libc: Access denied finding property "vendor.camera.aux.packagelist"

royalbhati, 2021-02-11:
Did you find any solution @lucastonon @tpharaoh ?