react-native-camera/react-native-camera #2788: Wrong cast methods used in the Windows implementation

qmatteoq, 2020-04-11:
# Bug Report

**To Do First**

- [ x ] Did you try latest release?
- [ x ] Did you try master?
- [ x ] Did you look for existing matching issues?

**Platforms**

Windows

**Versions**

- Windows: Windows 10 2004 (19041.173)
- react-native-camera: 3.22.1
- react-native: 0.61.5
- react: 16.9.0
- react-native-windows: 0.61.4

**Description/Current Behaviour**

1) Add react-native-camera to a React Native project with Windows support
2) Follow the [docs](https://react-native-community.github.io/react-native-camera/docs/installation) to link the Windows C++ module to the application
3) Open Visual Studio and deploy the React Native application
4) The solution fails to compile due to a series of errors related to the wrong method user for casting, like:

```
>C:\Users\mpagani\Source\React\CameraSample\node_modules\react-native-camera\windows\ReactNativeCameraCPP\ReactCameraView.cpp(67,67): error C4996: 'winrt::Microsoft::ReactNative::JSValue::Double': Use TryGetDouble, AsDouble, or AsJSNumber
>C:\Users\mpagani\Source\React\CameraSample\node_modules\react-native-camera\windows\ReactNativeCameraCPP\ReactCameraView.cpp(70,67): error C4996: 'winrt::Microsoft::ReactNative::JSValue::Double': Use TryGetDouble, AsDouble, or AsJSNumber
>C:\Users\mpagani\Source\React\CameraSample\node_modules\react-native-camera\windows\ReactNativeCameraCPP\ReactCameraView.cpp(73,68): error C4996: 'winrt::Microsoft::ReactNative::JSValue::Double': Use TryGetDouble, AsDouble, or AsJSNumber
>C:\Users\mpagani\Source\React\CameraSample\node_modules\react-native-camera\windows\ReactNativeCameraCPP\ReactCameraView.cpp(76,50): error C4996: 'winrt::Microsoft::ReactNative::JSValue::Boolean': Use TryGetBoolean, AsBoolean, or AsJSBoolean
>C:\Users\mpagani\Source\React\CameraSample\node_modules\react-native-camera\windows\ReactNativeCameraCPP\ReactCameraView.cpp(79,64): error C4996: 'winrt::Microsoft::ReactNative::JSValue::Double': Use TryGetDouble, AsDouble, or AsJSNumber
>C:\Users\mpagani\Source\React\CameraSample\node_modules\react-native-camera\windows\ReactNativeCameraCPP\ReactCameraView.cpp(93,46): error C4996: 'winrt::Microsoft::ReactNative::JSValue::String': Use TryGetString, AsString, or AsJSString
```
**Expected Behaviour**

The solution compiles without errors and the app is deployed

**Steps to Reproduce**

1) Add react-native-camera to a React Native project with Windows support
2) Follow the [docs](https://react-native-community.github.io/react-native-camera/docs/installation) to link the Windows C++ module to the application
3) Open Visual Studio and deploy the React Native application

**Additionals**

You can restore the project to a working state by following the suggestion included in the error message and replacing every method with the variant prefixed by As. For example, AsDouble(), AsBoolean(), AsString(), etc.

stmoy, 2020-04-14:
cc: @Kaiguo 

This looks like an upgrade breaking change in RN to 0.61. We need to update the module.

closed 2020-05-04 by ddalp

n1ru4l, 2020-05-05:
:tada: This issue has been resolved in version 3.24.0 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v3.24.0)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

PR #2802 (2020-05-04): Windows fixes

# Summary

Fixes #2788 
Fixes #2789 
Closes https://github.com/microsoft/react-native-windows/issues/4605

react-native-windows had some breaking API changes during 0.61 upgrade, updated JSValue casting to use the new API. Also made some changes to store pictures on disk by default and  updated parameter names to match TakePictureResponse fields.

## Test Plan

### What's required for testing (prerequisites)?

### What are the steps to reproduce (after prerequisites)?

## Compatibility

| OS      | Implemented |
| ------- | :---------: |
| iOS     |    ❌     |
| Android |    ❌   [... cut]