react-native-camera/react-native-camera #2789: Taking a picture in Windows throws an error

qmatteoq, 2020-04-11:
# Bug Report

**To Do First**

- [ x ] Did you try latest release?
- [ x ] Did you try master?
- [ x ] Did you look for existing matching issues?

**Platforms**

Windows

**Versions**

- Windows: Windows 10 2004 (19041.173)
- react-native-camera: 3.22.1
- react-native: 0.61.5
- react: 16.9.0
- react-native-windows: 0.61.4

**Description/Current Behaviour**

When you call the takePictureAsync() method exposed by the camera object, you get the following error on Windows 10:

```
:8081/index.bundle?platform=windows&dev=true&hot=false&inlineSourceMap=true:67975 Possible Unhandled Promise Rejection (id: 1):
Error: target parameter not specified.
Error: target parameter not specified.
    at Object.fn [as takePicture] (http://localhost:8081/index.bundle?platform=windows&dev=true&hot=false&inlineSourceMap=true:2844:36)
    at Camera.takePictureAsync$ (http://localhost:8081/index.bundle?platform=windows&dev=true&hot=false&inlineSourceMap=true:97672:65)
    at tryCatch (http://localhost:8081/index.bundle?platform=windows&dev=true&hot=false&inlineSourceMap=true:25709:19)
    at Generator.invoke [as _invoke] (http://localhost:8081/index.bundle?platform=windows&dev=true&hot=false&inlineSourceMap=true:25885:24)
    at Generator.prototype.<computed> [as next] (http://localhost:8081/index.bundle?platform=windows&dev=true&hot=false&inlineSourceMap=true:25752:23)
    at tryCatch (http://localhost:8081/index.bundle?platform=windows&dev=true&hot=false&inlineSourceMap=true:25709:19)
    at invoke (http://localhost:8081/index.bundle?platform=windows&dev=true&hot=false&inlineSourceMap=true:25785:22)
    at http://localhost:8081/index.bundle?platform=windows&dev=true&hot=false&inlineSourceMap=true:25815:13
    at tryCallTwo (http://localhost:8081/index.bundle?platform=windows&dev=true&hot=false&inlineSourceMap=true:27074:7)
    at doResolve (http://localhost:8081/index.bundle?platform=windows&dev=true&hot=false&inlineSourceMap=true:27238:15)
```

**Expected Behaviour**

The photo is captured and ready to be manipulated.

**Steps to Reproduce**

1) Create a new React Native project
2) Add Windows support
3) Add the react-native-camera module and follow [the docs](https://react-native-community.github.io/react-native-camera/docs/installation) to link the module
4) Replace the app.js content with the sample code [from the documentation](https://react-native-community.github.io/react-native-camera/docs/rncamera)
5) Launch the app on Windows 10
6) Notice that the camera preview is properly displayed
7) Press the Snap button 
8) Observe the operation failing with the error reported in the bug description

stmoy, 2020-04-14:
CC: @kaiguo

closed 2020-05-04 by ddalp

n1ru4l, 2020-05-05:
:tada: This issue has been resolved in version 3.24.0 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v3.24.0)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

PR #2802 (2020-05-04): Windows fixes

# Summary

Fixes #2788 
Fixes #2789 
Closes https://github.com/microsoft/react-native-windows/issues/4605

react-native-windows had some breaking API changes during 0.61 upgrade, updated JSValue casting to use the new API. Also made some changes to store pictures on disk by default and  updated parameter names to match TakePictureResponse fields.

## Test Plan

### What's required for testing (prerequisites)?

### What are the steps to reproduce (after prerequisites)?

## Compatibility

| OS      | Implemented |
| ------- | :---------: |
| iOS     |    ❌     |
| Android |    ❌   [... cut]