react-native-camera/react-native-camera #2791: onBarCodeRead() event returns wrong "bounds" coordinates

ceefour, 2020-04-12:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: 10
- iOS:
- react-native-camera: master 296045e
- react-native: 0.61.5
- react: 16.9.0
  

**Description/Current Behaviour**

Bounds is inaccurate. (most of the time?)

Example:

[image]

The coordinates don't make sense even mathematically, height is 1728 but there is a "y" point at 1878.5 ?

I've tried swapping x and y, no coordinates give the proper bounds.

**Expected Behaviour**

Bounds are accurate strictly within the coordinate system of width/height, no additional math and rotation should be needed by the `onBarCodeRead()` event handler.

**Steps to Reproduce**

`onBarCoreRead=event handler`

**Additionals**

I think it has something to do with left/top/scanWidth/scanHeight. Probably not the entire image is scanned, which means the bounds are relative to the scan bounds, not to entire image.

Related to #1880, #874, #2786.

[image]

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

fabriziobertoglio1987, 2020-06-15:
yes https://github.com/react-native-community/react-native-camera/pull/2766

fabriziobertoglio1987, 2020-06-15:
If you read the code from [zxing Android example](https://github.com/zxing/zxing/blob/a65631d0b643e778b8b1c1975e4f18f35e401fa4/android/src/com/google/zxing/client/android/camera/CameraManager.java#L319-L325), they include this logic to fix the Screen/Image ratio distorsions.

I hide the below comment https://github.com/react-native-community/react-native-camera/pull/2766#issuecomment-606023440

> Thanks @sibelius I am working with the react native hooks api to publish a simple solution.
> 
> @computerjazz @YLeaves @lededje @bqwang91 I found out that the Camera Resolution is [the quality used](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md#defaultvideoquality) and it does not relate to the `screen` resolution. I can preview/display the video at _1080p_ (1920x1080 pixel), but record at _720p_ (1280 x 720pixel). The previous approach below needs to be replaced with a method which uses the react native bridge to retrieve the current resolution from the camera.
> 
> ```js
> const CAM_VIEW_HEIGHT = Dimensions.get('screen').width * 1.5;
> const CAM_VIEW_WIDTH = Dimensions.get('screen').width;
> ```
> 
> This could be connected to the distortion created between the screen and recording resolution, which would change the crop coordinates..
> 
> The aspect ratio may also be involved in the calculation ...?
> How can I get the camera resolution from javascript without adding any method?
> 
> This is how `zxing` android app selects the correct resolution with the method [findBestPreviewSizeValue](https://github.com/zxing/zxing/blob/0cf3b9be71680f50c90a71ca26ce0d33664b0dd6/android-core/src/main/java/com/google/zxing/client/android/camera/CameraConfigurationUtils.java#L273-L345). The method is called to determine both `camera` and `screen` resolution in [`getFramingRectInPreview`](https://github.com/zxing/zxing/blob/a65631d0b643e778b8b1c1975e4f18f35e401fa4/android/src/com/google/zxing/client/android/camera/CameraManager.java#L256-L257).
> 
> An alternative I am taking in consideration is setting myself the `defaultResolution` with the existing api and then using that resolution, hopefully it will be the one used for scanning the QR codes... I need to read the Android sourcecode from Camera1, but I believe that we are using [CamCorder](https://developer.android.com/guide/topics/media/mediaplayer) profile to set the default profile for recording....
> 
> I believe that I should find the camera resolution for taking pictures and not videos, as it is an image being cropped and that may be a different process from Video Recording..
> 
> Thanks

sibelius, 2020-06-15:
if someone want to bring this functionally from zxing feel free to

we can review

the only way to make sure this work forever would be to add proper automated tests

samchan0221, 2021-11-18:
Seems like it is still not fixed? I am having the same issue.

chiubaka, 2021-11-19:
I think you'd be forced to use `onGoogleVisionBarcodesDetected` to use my fix, but I have a PR open with a fix for this for that method: https://github.com/react-native-camera/react-native-camera/pull/3381

From what I can see in the code, `onBarCodeRead` follows an entirely different pathway, and my fix likely hasn't affect it at all. However, the fix for that method is likely similar to what I've done in my PR--I'd guess that the reverse transformation for the coordinate systems was not calculated correctly.