react-native-camera/react-native-camera #2798: bugs revealed by on-device tests

wolverineks, 2020-04-18:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

android
* camera does not mount
  * workaround: `style={ height: 100, width: 100 }`

* back camera flashes / runs slow
  * workaround: `ratio={"16:9"}`

* camera not ready even after `onCameraReady`
  * workaround: `await waitFor(1000)`

pixel 2
* Starting video recording failed. Another recording might be in progress.

pixel3 xl
* unknown crash on tests
* unknown crash on advanced example
* doesnt run basic example (duplicate react-native in haste map)

**Versions**

- Android:
- iOS:
- react-native-camera:
- react-native:
- react:
  

**Description/Current Behaviour**

**Expected Behaviour**

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

wolverineks, 2020-04-18:
also, my previous pr broke all the tests, #2797 fixes them

cristianoccazinsp, 2020-04-23:
For android, is it Camera1 or Camera2 implementation? 

I'm curious about `back camera flashes / runs slow`: Changing the aspect ratio is definitely a not a good idea (for pictures at least). Is this for photos or videos? The issue is probably that you need to make sure that the width/height of the view matches your aspect ratio.

`camera not ready even after onCameraReady`: This one is also interesting. It should definitely fire at the right time on Android with Camera1

Pixel 2: Video works fine for me on a real device.

sibelius, 2020-04-23:
you can run tests locally on devices

check the code here https://github.com/react-native-community/react-native-camera/blob/master/examples/tests/tests/index.js

there are tests for both Camera1 and Camera2

how does we run tests @wolverineks ?

wolverineks, 2020-04-23:
the tests run automatically when the tests app loads

wolverineks, 2020-04-23:
* the android tests run for both camera apis
* pixel 2 fails on simulator, i dont have a pixel 2 device
* camera not ready:
https://github.com/react-native-community/react-native-camera/blob/48a3a4dfd86a3ca064cb7166a720d7ee7b716653/examples/tests/tests/photos/photos.js#L29

wolverineks, 2020-04-23:
flashing/slow camera
back camera api1 / api2

https://drive.google.com/open?id=1in7NhmFQ5-IvCmnWyao-kSO5YoA9ujso

wolverineks, 2020-04-23:
how do i match view and ratio?

wolverineks, 2020-04-23:
onCameraReady fires before camera is ready on pixel 3 device

sibelius, 2020-04-23:
ins't better run tests using jest/detox ?

wolverineks, 2020-04-23:
> ins't better run tests using jest/detox ?

what do you mean by "better"?

closed 2020-06-30 by stale[bot]