react-native-camera/react-native-camera #2799: how can I mute the prompt when record

original-curtain, 2020-04-20:
# Question

**To Do First**
- [ ] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [ ] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [ ] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**

**Tags**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

wolverineks, 2020-04-21:
i have the same gripe.
[heres](https://stackoverflow.com/questions/17041238/android-how-to-disable-sound-on-video-recording-start) and old stack overflow
havent tried it though

wolverineks, 2020-04-21:
also looks like a duplicate of #959

original-curtain, 2020-04-22:
> also looks like a duplicate of #959

thank you for your reply,but there is still no effective way to solve this problem.\orz

renatobenks, 2020-06-21:
what you mean with prompt @original-curtain? once it seems to be regarding a feature that already built as mentioned by @wolverineks, let me know what you need and how I can help you

closed 2020-08-28 by stale[bot]