react-native-camera/react-native-camera #2803: iOS: When trying to record audio in the background camera in video mode is not recording neither my voice or the bg audio

esmatnawahda1, 2020-04-25:
# Bug Report

**To Do First**

- [ ] Did you try latest release? yes
- [ ] Did you try master? yes
- [ ] Did you look for existing matching issues? yes

**Platforms**

The issue occurs only on iOS

**Versions**
"react-native-camera": "^3.23.1"

- Android:
- iOS:
- react-native-camera: 3.23.1
- react-native: 0.61.5
- react: 16.9.0
  
5.0.7

**Description/Current Behaviour**
Trying to run an audio while the video is recording.
on Android all is going well, but on iOS it is not working (video is recording without any audio)
```
<RNCamera
                    onRecordingStart={(recordAsync) => {
                            player = new Player(`file://${soundUri}`, { mixWithOthers: true }).play()
                    }}
/>
```

**Expected Behaviour**
We expect to record the voice with the video and be able to hear it after finishing when we play the video.

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

esmatptr, 2020-05-01:
@alloy any help here?

alloy, 2020-05-01:
I have no idea off-hand, this sounds like something somebody really needs to dig into. If you aren’t able to do that, you will either have to wait until somebody else does, or–as the issue template apparently mentions–you can put a bounty on it and pay somebody to do that work.

> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

closed 2020-07-08 by stale[bot]

SamiChab, 2021-02-16:
Check this! https://github.com/react-native-video/react-native-video/issues/2228#issuecomment-773709842