react-native-camera/react-native-camera #2804: Reease APK Error on React-Native 0.62.0

DanishWajid, 2020-04-25:
# Question
I have upgraded the app to react-native 0.62.0 from 0.59.10. In start app crashes, but I have added some code in android/build.gradle to resolve dependencies. Now the app works fine in development mode but gives error when I try to make APK.

I have added the following code in android/build.Gradle (in all projects section)

*android/build.gradle *
`// Top-level build file where you can add configuration options common to all sub-projects/modules.

buildscript {
    ext {
        buildToolsVersion = "28.0.3"
        minSdkVersion = 16
        compileSdkVersion = 28
        targetSdkVersion = 28
    }
    repositories {
        google()
        jcenter()
    }
    dependencies {
        classpath('com.android.tools.build:gradle:3.6.1')
        classpath 'com.google.gms:google-services:4.3.3'
        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}

allprojects {
    repositories {
        //start here
    configurations.all {
 resolutionStrategy.eachDependency { DependencyResolveDetails details ->
   def requested = details.requested
       if (requested.group == 'com.google.android.gms') {
          details.useVersion '12.0.1'
       }
       if (requested.group == 'com.google.firebase') {
          details.useVersion '12.0.1'
         }
       }
     }
    //end
        mavenLocal()

        google()
        jcenter()
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url("$rootDir/../node_modules/react-native/android")
        }
        maven {
            // Android JSC is installed from npm
            url("$rootDir/../node_modules/jsc-android/dist")
        }
        maven { url "https://jitpack.io" }
     
    }
}
`

Build Error:
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:lintVitalRelease'.
> Could not resolve all artifacts for configuration ':react-native-camera:mlkitDebugRuntimeClasspath'.
   > Could not find com.google.firebase:firebase-ml-vision:12.0.1.
     Searched in the following locations:
       - file:/C:/Users/Danish Wajid/.m2/repository/com/google/firebase/firebase-ml-vision/12.0.1/firebase-ml-vision-12.0.1.pom
       - https://dl.google.com/dl/android/maven2/com/google/firebase/firebase-ml-vision/12.0.1/firebase-ml-vision-12.0.1.pom
       - https://jcenter.bintray.com/com/google/firebase/firebase-ml-vision/12.0.1/firebase-ml-vision-12.0.1.pom
       - file:/E:/Projects/premise_hq_native/node_modules/react-native/android/com/google/firebase/firebase-ml-vision/12.0.1/firebase-ml-vision-12.0.1.pom
       - file:/E:/Projects/premise_hq_native/node_modules/jsc-android/dist/com/google/firebase/firebase-ml-vision/12.0.1/firebase-ml-vision-12.0.1.pom
       - https://jitpack.io/com/google/firebase/firebase-ml-vision/12.0.1/firebase-ml-vision-12.0.1.pom
       - https://repo.maven.apache.org/maven2/com/google/firebase/firebase-ml-vision/12.0.1/firebase-ml-vision-12.0.1.pom
     Required by:
         project :react-native-camera
   > Could not find com.google.firebase:firebase-ml-vision-face-model:12.0.1.
     Searched in the following locations:
       - file:/C:/Users/Danish Wajid/.m2/repository/com/google/firebase/firebase-ml-vision-face-model/12.0.1/firebase-ml-vision-face-model-12.0.1.pom
       - https://dl.google.com/dl/android/maven2/com/google/firebase/firebase-ml-vision-face-model/12.0.1/firebase-ml-vision-face-model-12.0.1.pom
       - https://jcenter.bintray.com/com/google/firebase/firebase-ml-vision-face-model/12.0.1/firebase-ml-vision-face-model-12.0.1.pom
       - file:/E:/Projects/premise_hq_native/node_modules/react-native/android/com/google/firebase/firebase-ml-vision-face-model/12.0.1/firebase-ml-vision-face-model-12.0.1.pom
       - file:/E:/Projects/premise_hq_native/node_modules/jsc-android/dist/com/google/firebase/firebase-ml-vision-face-model/12.0.1/firebase-ml-vision-face-model-12.0.1.pom
       - https://jitpack.io/com/google/firebase/firebase-ml-vision-face-model/12.0.1/firebase-ml-vision-face-model-12.0.1.pom
       - https://repo.maven.apache.org/maven2/com/google/firebase/firebase-ml-vision-face-model/12.0.1/firebase-ml-vision-face-model-12.0.1.pom
     Required by:
         project :react-native-camera

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

closed 2020-07-02 by stale[bot]

oussamajilal, 2020-07-22:
Having the same error with version `3.31.1` and react-native `0.59.8` and not using ML Kit (`missingDimensionStrategy 'react-native-camera', 'general'`)

Any idea how to fix this ?

chpriyanka433, 2021-08-19:
camera is not working in android 11 version and react-native version is 0.61 and build Tool Version is 30.0.2, min SDK version  30,

can you give me any idea how to fix it