react-native-camera/react-native-camera #2805: Attempt to read from field 'int android.media.CamcorderProfile.quality' on a null object reference

zhonghanwen, 2020-04-26:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android:
- react-native-camera: 3.23.1
  

**Description/Current Behaviour**

**Expected Behaviour**

**Steps to Reproduce**

```java
2020-04-26 19:25:19.369 10952-18845/com.ainirobot: E/CAMERA_1::: Record start failed
    java.lang.NullPointerException: Attempt to read from field 'int android.media.CamcorderProfile.quality' on a null object reference
        at com.google.android.cameraview.Camera1.setUpMediaRecorder(Camera1.java:1516)
        at com.google.android.cameraview.Camera1.record(Camera1.java:812)
        at com.google.android.cameraview.CameraView.record(CameraView.java:632)
        at org.reactnative.camera.RNCameraView.access$2601(RNCameraView.java:31)
        at org.reactnative.camera.RNCameraView$3.run(RNCameraView.java:303)
        at android.os.Handler.handleCallback(Handler.java:751)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:154)
        at android.os.HandlerThread.run(HandlerThread.java:61)
2020-04-26 19:25:19.377 10952-18813/com.ainirobot E/ReactNativeJS: , 'Starting video recording failed. Another recording might be in progress.'
```

**Additionals**
**My android Device is no Black type camera.**

closed 2020-07-03 by stale[bot]

kmcmanimen, 2020-10-08:
If anyone comes across this, the issue for us was related to a device that does not have a back facing camera.  When passing null the camera would throw an error and not proceed.  We solved the issue by patching the library to set the default if the profile is null:

```
diff --git a/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/Camera1.java b/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/Camera1.java
index 30a0cb2..6c5e5f3 100644
--- a/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/Camera1.java
+++ b/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/Camera1.java
@@ -1556,12 +1556,14 @@ class Camera1 extends CameraViewImpl implements MediaRecorder.OnInfoListener,
         mVideoPath = path;
 
         CamcorderProfile camProfile;
-        if (CamcorderProfile.hasProfile(mCameraId, profile.quality)) {
+        if (profile != null && CamcorderProfile.hasProfile(mCameraId, profile.quality)) {
             camProfile = CamcorderProfile.get(mCameraId, profile.quality);
         } else {
             camProfile = CamcorderProfile.get(mCameraId, CamcorderProfile.QUALITY_HIGH);
         }
-        camProfile.videoBitRate = profile.videoBitRate;
+        if (profile != null) {
+          camProfile.videoBitRate = profile.videoBitRate;
+        }
         setCamcorderProfile(camProfile, recordAudio, fps);
 
         mMediaRecorder.setOrientationHint(calcCameraRotation(mOrientation != Constants.ORIENTATION_AUTO ? orientationEnumToRotation(mOrientation) : mDeviceOrientation));
diff --git a/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/Camera2.java b/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/Camera2.java
index 6a0500f..0e9070d 100644
--- a/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/Camera2.java
+++ b/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/Camera2.java
@@ -1381,10 +1381,12 @@ class Camera2 extends CameraViewImpl implements MediaRecorder.OnInfoListener, Me
         mVideoPath = path;
 
         CamcorderProfile camProfile = profile;
-        if (!CamcorderProfile.hasProfile(Integer.parseInt(mCameraId), profile.quality)) {
+        if (profile != null && !CamcorderProfile.hasProfile(Integer.parseInt(mCameraId), profile.quality)) {
             camProfile = CamcorderProfile.get(CamcorderProfile.QUALITY_HIGH);
         }
-        camProfile.videoBitRate = profile.videoBitRate;
+        if(profile != null) {
+          camProfile.videoBitRate = profile.videoBitRate;
+        }
         setCamcorderProfile(camProfile, recordAudio);
 
         mMediaRecorder.setOrientationHint(getOutputRotation());
```

Best of luck!

piccinnigius, 2021-01-13:
@kmcmanimen not working for me :/