react-native-camera/react-native-camera #2806: recordAsync() output files having different orientation while using front and back camera

arjun-kava, 2020-04-26:
Hello, guys

I am using it on Android device. I am implementing recording pause with front and back camera switch. I found that both of the camera setting different orientation.   
```js
.recordAsync({
            orientation: "portrait",
            quality: RNCamera.Constants.VideoQuality["1080p"],
            skipProcessing: false,
})
````

As below in metadata. first response is of back camera while second is of front camera. 
Metadata 
```json
[
  {
    "audio_Mime": "audio/mp4a-latm",
    "data": "{max-bitrate=96000, sample-rate=48000, track-id=2, file-format=video/mp4, mime=audio/mp4a-latm, bitrate=96000, language=, aac-profile=2, durationUs=7240354, aac-format-adif=0, channel-count=2, max-input-size=65538, csd-0=java.nio.HeapByteBuffer[pos=0 lim=2 cap=2]}",
    "duration": "7264",
    "filepath": "/data/user/0/com.xyz/cache/Camera/f42216ad-5847-40b8-b5ea-f934347b82d4.mp4",
    "frame_count": "218",
    "frame_rate": "0.000000",
    "has_Audio": "yes",
    "height": "1080",
    "orientation": "90",
    "width": "1920"
  },
  {
    "audio_Mime": "audio/mp4a-latm",
    "data": "{max-bitrate=96000, sample-rate=48000, track-id=2, file-format=video/mp4, mime=audio/mp4a-latm, bitrate=96000, language=, aac-profile=2, durationUs=4842583, aac-format-adif=0, channel-count=2, max-input-size=65538, csd-0=java.nio.HeapByteBuffer[pos=0 lim=2 cap=2]}",
    "duration": "4912",
    "filepath": "/data/user/0/com.xyz/cache/Camera/d48a2304-8765-4c77-b91f-850381dd4c3a.mp4",
    "frame_count": "147",
    "frame_rate": "0.000000",
    "has_Audio": "yes",
    "height": "1080",
    "orientation": "270",
    "width": "1920"
  }
]
```

I did for through library code, if I could contribute for the same but didn't found exact code for the same. Did anyone faced the same issue?

closed 2020-07-03 by stale[bot]

dev3288, 2021-02-04:
@arjun-kava Were you able to solve the problem ?

johnylawrence1987, 2022-04-13:
Pump