react-native-camera/react-native-camera #2807: Play background audio playing while taking video

nica0012, 2020-04-26:
Is it possible to have audio like music playing in the background while recording a video, like Snapchat for example?

arjun-kava, 2020-04-27:
Hey, I think that you can merging audio file in video would work. If you are not dealing with heavy videos. You can use [react-native-ffmpeg](https://www.npmjs.com/package/react-native-ffmpeg). Otherwise do it on server.

nica0012, 2020-04-27:
@arjun-kava Hey Arjun, thanks for the reply. I am looking for something not as advanced as adding audio directly to the video. A common use case would be if for example the user is listening to a song on their phone from the music app, now on my app when they record a video or open the camera the music will stop playing until they leave the camera. I would like the music to remain playing even when they record so they can still hear the music audio in the video. I hope thats clear :)

cristianoccazinsp, 2020-05-02:
If all you want is to continue music playback, then use `captureAudio={false}`. That way the video/camera won't request audio access.

If you want to listen to music AND capture video with audio, then that's currently not possible because the audio session requests exclusive access.
It may be possible to request "mixed" access to the audio session, but that usually results in poor recording quality because the device does some changes to the microphone (selects worse microphones, etc.)

See my answer here; similar issue: https://github.com/react-native-community/react-native-camera/issues/2813

nica0012, 2020-05-03:
@cristianoccazinsp Thanks! captureAudio={false} would be the perfect functionality if it recorded the audio too. I am interested in requesting mixed access, are there any examples of this besides the links you send on the other issue? I don't have much experience with Swift. Cheers!

cristianoccazinsp, 2020-05-03:
I don't have any examples, and it is not implemented in the library right
now. However, it should be simple to implement with the info I shared.

nica0012, 2020-05-03:
@cristianoccazinsp Yep you're right, I got it to work :) Thanks!

nica0012, 2020-05-03:
Hey @cristianoccazinsp - I’ve updated my issue with some more information, any idea why this could be?

cristianoccazinsp, 2020-05-03:
@nica0012 maybe this line here is related: https://github.com/react-native-community/react-native-camera/blob/master/ios/RN/RNCamera.m#L1428

Basically, the audio session is "returned" so other background audio sessions can resume. 

Otherwise, I would look at the debugger/logs to see if something else is interfering.

nica0012, 2020-05-04:
@cristianoccazinsp Thank you - I have been messing with code in that area for quite some time now... It seems to be stopping the background audio session not only when you return back to the camera, but when the camera updates in any way (like when the camera is flipped)... The debug logs don't give much feedback unfortunately. I'm going to keep picking at it - if you have any other suggestions i'd appreciate it!

lucastonon, 2020-05-25:
I am having the same problem

nica0012, 2020-05-25:
@lucastonon I still haven't figured it out unfortunately. It is only working for the first attempt. I wish this was implemented in r-n-c instead of having to mess around with the native code with risks of breaking the whole thing, or at-least documented somewhere. I will let you know if I end up figuring this out!

closed 2020-08-01 by stale[bot]