react-native-camera/react-native-camera #2810: After record async video file is not exist at temp folder

DevAelius, 2020-04-28:
# Bug Report

**To Do First**

 Did you try latest release = yes
 Did you try master = yes

**Platforms**
-Linux mint 

**Versions**

- Android: oreo in emulator pixel 2 
- iOS: n/a
- react: "16.9.0",
- react-native: "0.61.5",
- react-native-camera: "^3.23.1",

**Description/Current Behaviour**
i want record video with rn camera  with below code

```
async startRecording() {
 
   await this.camera.recordAsync({
      maxDuration: 30,
    }).then(data => {
      console.log('data of startRecording', data)
      this.saveVideo(data)
    }).catch(e => {
      console.log('catch of startRecording', e)
    })
}
```

**it will return 4 things** 

1 uri
2 deviceOrientation
3 videoOrientation
4 isRecordingInterrupted

After that i want to move recorded video file from below location to **download dir location**

**Path of recorded video which receive from data.uri**
`file:///data/user/0/com.videoapp/cache/Camera/b5ba0d8f-07ea-4755-8f7f-a88d5cc7a331.mp4"`

**My Code for moving file** 

 ```
saveVideo = async (data) => {
    try {
      RNFetchBlob.fs.mv(data.uri, RNFetchBlob.fs.dirs.DownloadDir)
        .then((e) => {
          console.log('MSG =>in RNFetchBlob  ', e)
        })
        .catch(e => { console.log('MSG =>in RNFetchBlob  ', e) })
    } catch (e) {
      console.log('MSG => Main Catch', e)
    }
  }
```

After all this it is throw error =>
**Source file at path does not exist**

**Expected Behaviour**

1. Recorded video should **properly save at temp location** which i receive from **data.uri**
     so i can move video to other path.

**Steps to Reproduce**

I take reference from below link 

[https://medium.com/@masochist.aman/capturing-images-with-react-native-203e24f93eb9](url) 

I also save URI in state 
```
async startRecording() {
    this.setState({ recording: true });

    await this.camera.recordAsync({
      maxDuration: 30,
    }).then(data => {
      console.log('data of startRecording', data)
      
      this.setState({
        videoPath: data.uri
      })
    }).catch(e => {
      console.log('catch of startRecording', e)
    })

}

```

and then =>

```
<Video
source={{uri:this.state.videoPath}}
style={{height:300,width:300}}
/>

```

cristianoccazinsp, 2020-05-02:
Don't move the file. Just set the record settings to save the file to your final destination already. I think your issue is that you are moving a file to a folder. You should define the target file name as well instead of just the file.

Set the `path` option when calling `recordAsync`. You must set a file name and extension, in addition to the folder.

closed 2020-07-09 by stale[bot]

shubham696, 2020-12-30:
> # Bug Report
> **To Do First**
> 
> Did you try latest release = yes
> Did you try master = yes
> 
> **Platforms**
> -Linux mint
> 
> **Versions**
> 
> * Android: oreo in emulator pixel 2
> * iOS: n/a
> * react: "16.9.0",
> * react-native: "0.61.5",
> * react-native-camera: "^3.23.1",
> 
> **Description/Current Behaviour**
> i want record video with rn camera with below code
> 
> ```
> async startRecording() {
>  
>    await this.camera.recordAsync({
>       maxDuration: 30,
>     }).then(data => {
>       console.log('data of startRecording', data)
>       this.saveVideo(data)
>     }).catch(e => {
>       console.log('catch of startRecording', e)
>     })
> }
> ```
> 
> **it will return 4 things**
> 
> 1 uri
> 2 deviceOrientation
> 3 videoOrientation
> 4 isRecordingInterrupted
> 
> After that i want to move recorded video file from below location to **download dir location**
> 
> **Path of recorded video which receive from data.uri**
> `file:///data/user/0/com.videoapp/cache/Camera/b5ba0d8f-07ea-4755-8f7f-a88d5cc7a331.mp4"`
> 
> **My Code for moving file**
> 
> ```
> saveVideo = async (data) => {
>    try {
>      RNFetchBlob.fs.mv(data.uri, RNFetchBlob.fs.dirs.DownloadDir)
>        .then((e) => {
>          console.log('MSG =>in RNFetchBlob  ', e)
>        })
>        .catch(e => { console.log('MSG =>in RNFetchBlob  ', e) })
>    } catch (e) {
>      console.log('MSG => Main Catch', e)
>    }
>  }
> ```
> 
> After all this it is throw error =>
> **Source file at path does not exist**
> 
> **Expected Behaviour**
> 
> 1. Recorded video should **properly save at temp location** which i receive from **data.uri**
>    so i can move video to other path.
> 
> **Steps to Reproduce**
> 
> I take reference from below link
> 
> [https://medium.com/@masochist.aman/capturing-images-with-react-native-203e24f93eb9](url)
> 
> I also save URI in state
> 
> ```
> async startRecording() {
>     this.setState({ recording: true });
> 
>     await this.camera.recordAsync({
>       maxDuration: 30,
>     }).then(data => {
>       console.log('data of startRecording', data)
>       
>       this.setState({
>         videoPath: data.uri
>       })
>     }).catch(e => {
>       console.log('catch of startRecording', e)
>     })
> 
> }
> ```
> 
> and then =>
> 
> ```
> <Video
> source={{uri:this.state.videoPath}}
> style={{height:300,width:300}}
> onError={(e)=>{console.log(e)}} <-- an Errors says that try to load empty video source
> />
> ```

@DevAelius  did you found the solution for this

vkachan, 2021-01-12:
> @DevAelius did you found the solution for this

Hey @shubham696  just encountered with the same issue here. Shame that devs can't say anything how they had solved the issue. In my case I just replaced 'file://' with empty string in the data.uri and then you can copy the video file. Hope it helps.