react-native-camera/react-native-camera #2811: mirrorVideo not work on Android

kperreau, 2020-04-28:
# Bug Report

**To Do First**

- [ x] Did you try latest release?
- [ x] Did you try master?
- [ x] Did you look for existing matching issues?

**Platforms**
Android

**Versions**

- Android: 9
- react-native-camera: 136
- react-native: 0.62.2
- react: 16.11.0
- react-navigation: 4.3.8

**Description/Current Behaviour**

Option `mirrorVideo` from `recordAsync()` not work on Android (recording video).

**Expected Behaviour**

Video should be mirrored.

**Steps to Reproduce**

Record a video on Android device with option `mirrorVideo` set to `true`.

fabriziobertoglio1987, 2020-05-02:
search before posting issue https://github.com/react-native-community/react-native-camera/issues/2594#issuecomment-555044291

joseym90, 2020-06-26:
I have the same issue,

Android: 10
react-native-camera: 3.30.0
react-native: 0.61.4
react: 16.9.0
react-navigation: 4.0.10

The mirrorVideo option doesn't make any effect on the recorded video. With the back-facing camera, video is normally recorded (as same as you see during the process). With the front camera, the recorded video is mirrored respecting the recorded image.

unandreshevia, 2020-09-02:
I'm having the exact same issue here.

Android: 9
react-native-camera: 3.30.0
react-native: 0.59.5

closed 2021-01-13 by stale[bot]