react-native-camera/react-native-camera #2813: Play audio in headphones while recording video and microphone

tfsjohan, 2020-04-30:
# Question

**To Do First**
- [x] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [x] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [x] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**
I want to record video and sound of people singing to a song provided by the application. I wan't the song to play in headphones, while recording video and sound from microphone. 

Most video recorders pause audio and get exclusive access to audio. I wonder if this library has any kind of support for keep playing audio while recording. 

I do not want to merge the played audio into the video recording. Just the microphone and video. 

**Tags**
audio video record microphone

nica0012, 2020-05-01:
I am also looking for the same behaviour, but instead without headphones (background audio to be played while taking video from the device speakers)

fabriziobertoglio1987, 2020-05-02:
playing a sound and recording a video are 2 different functionalities... but you can research in the android sdk docs and also in the existing react-native libs

https://www.npmjs.com/package/react-native-sound-player

cristianoccazinsp, 2020-05-02:
I recall iOS being quite picky with recording audio while also playing audio. Right now it's one or the other, but not both, but iOS has some "audio settings" to allow multiple access to the recording/playback devices. However, when I tested it, audio recording ended up with very poor quality.

For iOS, you should look around these lines: https://github.com/react-native-community/react-native-camera/blob/master/ios/RN/RNCamera.m#L1366
You are looking for this setting here: https://developer.apple.com/documentation/avfoundation/avaudiosessioncategoryplayandrecord?language=objc

For Android, it seems like it is already supported. I am able to have background music from my speakers while recording video with RNCamera and `captureAudio=true`. 

The update for iOS could be as simple as adding a new iOS only property (e.g., `mixedAudio` or `playAndRecord` and setting the above value on the audio session configuration.)

tfsjohan, 2020-05-02:
Thank you! I think I'll try to fork it and see how it works. I was most afraid of Android actually, since I mostly have done native development for iOS, so I'm glad you say it might already be working for Android.

closed 2020-05-02 by tfsjohan

NooruddinLakhani, 2020-05-06:
My issue is, when I start the video recording by putting headphones, the music plays in the background/headphone but doesn't play in recorded video. How would I achieve this?

nica0012, 2020-05-06:
@NooruddinLakhani From the sounds of it, if you have headphones playing music you will not be able to get the audio because the microphone can't detect the sound. react-native-sound might be a good option? 

If you play the audio without headphones does it work while recording?

NooruddinLakhani, 2020-05-07:
@nica0012 I don't think so "react-native-sound" works because there is no option to record sound with video recording.

Yes, without headphones, It records the sound playing with "react-native-sound" from the speaker

morganruffner, 2020-07-12:
> My issue is, when I start the video recording by putting headphones, the music plays in the background/headphone but doesn't play in recorded video. How would I achieve this?

This is actually what I would like to achieve, how were you able to do this?

NooruddinLakhani, 2020-10-07:
I recorded the video separately by muting a voice and record audio separately through [react-native-audio-player-recorder](https://github.com/Shinetechchina/react-native-audio-player-recorder#readme) library which record voice with mic and music playing in headphone then merge both to get the final video by FFmpeg library.