react-native-camera/react-native-camera #2816: FPS supported but not documented?

PanMan, 2020-05-01:
# Question
It seems to me FPS is supported as a RecordingOptions: 
https://github.com/react-native-community/react-native-camera/blob/master/src/RNCamera.js#L230
but not documented? Couldn't find anything in the docs about it. 
Is there a reason for this, or should it "just work"? Then I'll do a pull req for the docs. Thanks!

**To Do First**
- [X] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [X] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [X] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Tags**

camera, fps, ios

fabriziobertoglio1987, 2020-05-02:
Work in Progress

closed 2020-07-09 by stale[bot]

AkashPushkar, 2020-10-27:
Is this feature (fps) supported . If yes, I could not find it in the documentation ([https://react-native-camera.github.io/react-native-camera/docs/rncamera#recordasync-options-promise]). Is there a way to capture video at specified FPS (low value like 5fps).

ainnotate, 2022-07-13:
Hi, I'm looking for a custom FPS as well, 

Could someone update on the status, please?

Thank you.