react-native-camera/react-native-camera #2820: Is it possible to make filters with react-native-camera?

felire, 2020-05-04:
# Question

That's my question. Is it possible? Or should I implement a Native solution? Is it possible to capture the image in each moment and apply filters from javascript?

moustaphadiakhate, 2020-05-05:
Try to use Opencv in react native there have some libs like npm React-native-opencv3 it's be possible to apply filters here, it is the best solution i know if you don't want to write Native code.
am using this libs... Note that you don't get the frames in to your js part but it sould be enough to to the job of applying filters form camera frames.

sandys, 2020-06-11:
@moustaphadiakhate could you explain this part ? we are looking to build face filters as well.

renatobenks, 2020-06-12:
what kind of filter are you looking to use @sandys?

by the way, guys @moustaphadiakhate @felire, could you describe better which kinds of filters are you would like to use in the rn-camera.

because, as I see the rn-camera purpose, the only filters that MAYBE, could be available to use, would be the native ones, from the own device, but I'm really don't know if these filters would be available for us from in the device's OS, but probably, and I’m quite a convinced of that, this will not be available to be used.

in the case of you guys are looking for any other kind of support to use custom filters instead, would be pretty interesting, and I would love to hear, the ideas that you guys have about how should be this support.

either way, whether you guys are just looking for any filter to be used on, maybe, the white-balance feature would be useful for you guys, just take a look into that https://github.com/react-native-community/react-native-camera/pull/2865, and let me know if this works for all you.

felire, 2020-06-13:
@renatobenks A was talking about filters like sharing, postprocessing the image catching some pixels and detecting the hear for example and modifying you hear color, etc. Instagram filters for example

renatobenks, 2020-06-13:
I see your point on that, and it sounds plausible to be built, in camera use-cases, but not to be built-in at the API top-level. I mean, doesn’t make sense for us to build these filters already and just, expose them to be used, but instead, it would make a lot of sense to be exposed, enabling the filters be built, in your own uses-cases, in the way that you want, and the camera be able to help you with that.

The main point is that filters itself shouldn’t be a concern of camera, but we can build a great API to enable our users build that too easy as possible. 

So, in advance, can you share which filters do you need for now? Because it seems to me to be more practical to think based on your use-cases.

Did you look for white-balance feature? It will not gonna help you at all?

felire, 2020-06-13:
I really don't have to build any filter. It was just a doubt and it will be nice that this library have the possibility to build customisable filters and that stuff. It would be great that it has something to process pixels or something like that. But thanks you so much for taking the time to answer me