react-native-camera/react-native-camera #2821: Access cameras types with getCameraIdsAsync returns partial results

asaf-demetria, 2020-05-05:
# Bug Report

**To Do First**

- [X] Did you try latest release?
- [X] Did you try master?
- [X] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: 9
- iOS:
- react-native-camera: 3.23.1
- react-native: 0.62.0
- react: 16.11.0

**Description/Current Behaviour**

I'm using a [Xiaomi mi note 10](https://www.gsmarena.com/xiaomi_mi_note_10-9936.php) with the 108MP camera. When asking to get all available cameras, using the ```getCameraIdsAsync``` method, it returns only 2 options - the front camera and the default back camera (which is not the 108MP camera).

**Expected Behaviour**

I would expect getting at least 3 camera ids (1 for front and 2 for back).

**Steps to Reproduce**

```
<RNCamera
   ref={this.camera}
   onCameraReady={this.onCameraReady}
   style={styles.cameraPreview}
   type={RNCamera.Constants.Type.back}
   cameraId={cameraId}
   captureAudio={false}
   useCamera2API={true} /* Tried with and without the camera2 API option */
/>

onCameraReady = () => {
    if (this.state.availableCameras.length === 0) {
      this.camera.current.getCameraIdsAsync().then((availableCameras) => {
        if (availableCameras && availableCameras.length > 0) {
          this.setState(
            {
              availableCameras,
            },
            () => {
              this.updateCameraPictureSize();
            },
          );
        }
      });
    } else {
      this.updateCameraPictureSize();
    }
  };
```

asaf-demetria, 2020-05-07:
@sibelius If I have a physical access to such device, is there any relevant info I can get that will help investigate the issue?

sibelius, 2020-05-07:
add Log.d (console.log of java android) to discover what is wrong

closed 2020-07-13 by stale[bot]