react-native-camera/react-native-camera #2826: n.checkVideoAuthorizationStatus is not a function.

edwardchen, 2020-05-06:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

iOS

**Versions**

- Android:
- iOS:
- react-native-camera: 3.23.1
- react-native: 0.61.5
- react: 16.9.0
  

**Description/Current Behaviour**

Getting this error in sentry on production:
n.checkVideoAuthorizationStatus is not a function. 

**Expected Behaviour**

Expect to not get this error.

**Steps to Reproduce**

I'm unable to reproduce. Is anybody else seeing this error?

sibelius, 2020-05-06:
your linking is wrong

edwardchen, 2020-05-06:
It mostly works though. I just get random errors.

sibelius, 2020-05-06:
try to run examples

edwardchen, 2020-05-06:
I'm not sure what you mean. I have it already working in production.

I'm just seeing these random errors and was wondering if it anyone else saw the same thing. That means it's breaking for a small percentage of users.

aibrahim3546, 2020-05-08:
This issue is still occuring. I dont think my linking is wrong. Because im using react native > 0.60.
It's auto link. any idea why this is occuring?

aibrahim3546, 2020-05-10:
I solved this issue by putting `pod 'react-native-camera', path: '../node_modules/react-native-camera'` in the Podfile. I am using react native `v0.61.4`.

I refered to this link here
https://github.com/react-native-community/react-native-camera/issues/2275#issuecomment-570784831

closed 2020-07-17 by stale[bot]