react-native-camera/react-native-camera #2828: Barcode scanning sometimes return results even when no barcode is present in the vicinity.

AnurajDev, 2020-05-07:
I am using RNCamera for barcode scanning. Although scanning works good, sometimes the camera returns random tag values even when there are no barcode is present in the vicinity. 

Tried both onBarCodeRead and onGoogleVisionBarcodesDetected for capturing the result, but the problem persisted. 

Any leads towards a solution is much appreciated. 

> react: 16.9.0
> react-native: 0.61.5
> react-native-camera: 3.19.1

manishkumar03, 2020-06-05:
Did you try specifying the exact type of barcode  to be scanned using `googleVisionBarcodeType `? If you know in advance what types of barcodes you are looking for, this setting will avoid spurious results.

tusharmutreja200, 2020-06-17:
+1, I am also facing this issue. I think **googleVisionBarcodeType** prop must have the functionality to pass an array of barcode types

Udbhav12, 2020-06-17:
I am facing the exact same issue

vikas-singh-fareye, 2020-06-17:
facing same issue

sarinashish, 2020-06-17:
I am also facing this issue

closed 2020-08-23 by stale[bot]

dragma, 2021-11-29:
This bug is really there and quite annoying....

mrigo, 2021-12-15:
Hey, this bug is still in production! Anybody with a solution?

leoamato10, 2021-12-28:
same problem! Only on Android.