react-native-camera/react-native-camera #2830: Add Bar Code Scanning Support for React Native Windows Cpp/WinRT

woodrufs, 2020-05-10:
# Feature-Request: RNW Cpp/WinRT Bar Code Support

**Describe the Feature**
The prior, C#, implementation for React Native Windows supported barcode scanning. The new Cpp/WinRT implementation that was added in PR [Add Cpp/WinRT camera module support for React Native Windows](https://github.com/react-native-community/react-native-camera/pull/2723) removes this functionality.
To quote the PR,

> It supports most of functionalities what current CSharp library does, with one exception that bar code reader has not been implemented yet. We will leave the CSharp library in the depot for now(with some simple change so it works with current RNC again). The plan is to add the code reader support (pending integrating with a C++ version of ZXing) to C++/WinRT library then delete CSharp library completely.

**Possible Implementations**
Much easier said than done, but @ddalp mentions in his PR integrating with a C++ version of ZXing to achieve this functionality.

woodrufs, 2020-05-10:
The current C# implementation does not support RNW v.61. If the current C# implementation was updated to work with v.61, I would be very grateful and would not have the urgent need for a Cpp/WinRT bar code scanning implementation for v.61. Thank you to anyone who can help me with this!

asklar, 2020-05-10:
I wonder if we could leverage the [BarcodeScanner](https://docs.microsoft.com/en-us/uwp/api/Windows.Devices.PointOfService.BarcodeScanner?view=winrt-18362) WinRT class (see sample [here](https://github.com/Microsoft/Windows-universal-samples/tree/master/Samples/BarcodeScanner)). I'm not really familiar with it so I don't know if this requires a physical barcode scanner or if it is able to work with any camera.

woodrufs, 2020-05-10:
@asklar,

Thanks for such a quick reply (especially on a Sunday)! I will definitely do that, and I will get back to you this week when I have some time to try this.

woodrufs, 2020-05-12:
Hey @asklar,
I believe this should work. I used the sample you provided, and it read bar codes in various symbologies just fine. I don't have access to all the tools I wish I did granted I am working from home, so I was only able to test using my local machine and my webcam. I think this should be fine, though.

At first, I had a little trouble figuring out why it was not decoding some popular bar code symbologies. Instead of seeing the value I expected, I would see `"Decoded data unavailable. Raw label data: "`. After digging into the code, I noticed it was because the sample app didn't provide an exhaustive list of all bar code symbologies that can be decode. They were kind enough to leave a comment to make it easy to debug in [DataHelpers.h](https://github.com/microsoft/Windows-universal-samples/blob/master/Samples/BarcodeScanner/cpp/DataHelpers.h#L78-L92) 😃. 
After adding the ones I had issues with to the list, it worked great. Thought I would make a note to save a couple minutes for the next person who might run into it.

asklar, 2020-05-13:
Glad to hear it works, @woodrufs! Maybe you'd be able to modify the current Windows RNCamera module to add the functionality you need? :)

woodrufs, 2020-05-13:
@asklar,
Maybe I can try and take another swing at creating a native UI component this weekend when I am free. I _really_ struggled though previously. I am newer to UWP than I am RN (I know, sad), so I am doubting myself.

I just heard that we (Plex) are being demoed at Build next week. That kind of surprised me being the current state of RNW and how Plex is consuming RNW v.59 ☹️. I _really_ want to be that success story which is why I have been bothering you guys lately. I hope you understand. If no one at Microsoft can help me with this, I will try and see what I can do on my free time. 

PS - This package's description in `package.json` reads:
> "A Camera component for React Native. Also reads barcodes."

Without reading [PR 2723](https://github.com/react-native-community/react-native-camera/pull/2723), people are going to be under the assumption that bar codes can be scanned with RNW. Especially when they follow the directions Microsoft provides to get started with React Native.

asklar, 2020-05-13:
> @asklar,
> Maybe I can try and take another swing at creating a native UI component this weekend when I am free. I _really_ struggled though previously. I am newer to UWP than I am RN (I know, sad), [...]

agree, it looks like we should have updated the package.json when we dropped that suppport.

woodrufs, 2020-05-18:
@asklar,
Just came across this session for MS Build this week. Curious what is going to be said.. https://mybuild.microsoft.com/sessions/b1306a79-b43b-43be-bd59-460b8db0c7a8?source=sessions

jonthysell, 2020-07-23:
It looks like there's a C++ port of zxing here with winrt nuget packages: https://github.com/nu-book/zxing-cpp

closed 2020-08-11 by sibelius

n1ru4l, 2020-08-11:
:tada: This issue has been resolved in version 3.36.0 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v3.36.0)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

PR #2930 (2020-08-11): feat(windows): added barcode scanning support
# Summary

This PR updates the ReactNativeCameraCPP windows implementation with barcode scanning support via ZXing.

Closes #2830

Build:
* Fixed issue with edit and continue debug symbols
* Fixed ReactNativeCameraCPP61 builds
* Added dependency on huycn.zxingcpp.winrt for barcode support

ReactNativeCameraCPP RNCamera component:
* Added onBarCodeRead event support
* Added barCodeScannerEnabled property support
* Added barCodeTypes property support
* Added barCodeReadIntervalMS property to alter how often the scan
  occurs when enabled

ReactNative [... cut]
commit 2782a24d (2020-08-11): feat(windows): added barcode scanning support (#2930)

Build:
* Fixed issue with edit and continue debug symbols
* Fixed ReactNativeCameraCPP61 builds
* Added dependency on huycn.zxingcpp.winrt for barcode support

ReactNativeCameraCPP RNCamera component:
* Added onBarCodeRead event support
* Added barCodeScannerEnabled property support
* Added barCodeTypes property support
* Added barCodeReadIntervalMS property to alter how often the scan
  occurs when enabled

ReactNativeCameraCPP RNCamera module:
* Added BarCodeType constants

Other:
* Fixed intermittent issue with thread marshalling
* Re-r [... cut]