react-native-camera/react-native-camera #2832: Huawei P 30 Pro crash recording video

lfiorini, 2020-05-11:
# Bug Report

**To Do First**

- [ ] Did you try latest release?
- [ ] Did you try master?
- [X] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: 10
- react-native-camera: 3.23.1
- react-native: 0.61.5
- react: 16.9.0
  

**Description/Current Behaviour**

The app has been successfully tested on several devices like Pixel 4, XIAOMI Note 8 Pro, Huawei 20 Pro 20, ...  but in Huawei P 30 Pro the video recording crash after few seconds with preview freeze

**Additionals**

...
log1_2:05-11 17:40:17.734   872  1657 I ImagingSystem: #Camera# [263385508]< doAllocate_alloc_Count(0)_WxH(0x1)
log1_2:05-11 17:40:17.734   872  1657 I ImagingSystem: #Camera# [263385766]> doAllocate_alloc_Count(7)_WxH(233984x4)
log1_2:05-11 17:40:17.736   872  1657 I ImagingSystem: #Camera# [263385766]< doAllocate_alloc_Count(7)_WxH(233984x4)
log1_2:05-11 17:40:17.737   872  1657 I ImagingSystem: #Camera# [263391256]> doAllocate_alloc_Count(36)_WxH(480x180)
log1_2:05-11 17:40:17.743   872  1657 I ImagingSystem: #Camera# [263391256]< doAllocate_alloc_Count(36)_WxH(480x180)
log1_2:05-11 17:40:17.744   872  1657 I ImagingSystem: #Camera# [263410119]> doAllocate_alloc_Count(36)_WxH(720x2)
log1_2:05-11 17:40:17.745   872 20952 I DeviceBaseImpl: [HWA_CAM3] getSensorModuleSpec() getSensorModuleSpec cameraID: 2
log1_2:05-11 17:40:17.745   872 20952 I DeviceBaseImpl: [HWA_CAM3] getSensorModuleSpec() getSensorModuleSpec cameraID: 16
log1_2:05-11 17:40:17.745   872 20952 I DeviceBaseImpl: [HWA_CAM3] getSensorModuleSpec() getSensorModuleSpec cameraID: 2
log1_2:05-11 17:40:17.745   872 20952 I DeviceBaseImpl: [HWA_CAM3] getSensorModuleSpec() getSensorModuleSpec cameraID: 2
log1_2:05-11 17:40:17.745   872 20952 I DeviceBaseImpl: [HWA_CAM3] getSensorModuleSpec() getSensorModuleSpec cameraID: 16
log1_2:05-11 17:40:17.747   872  1657 I ImagingSystem: #Camera# [263410119]< doAllocate_alloc_Count(36)_WxH(720x2)
log1_2:05-11 17:40:17.747   872  1657 I ImagingSystem: #Camera# [263420030]> doAllocate_alloc_Count(8)_WxH(16000x1)
log1_2:05-11 17:40:17.748   872  1657 I ImagingSystem: #Camera# [263420030]< doAllocate_alloc_Count(8)_WxH(16000x1)
log1_2:05-11 17:40:17.748   872  1657 I ImagingSystem: #Camera# [263374036]< pipeline_start
log1_2:05-11 17:40:17.743   872   872 W pl#3    : type=1400 audit(0.0:23121): avc: denied { read } for pid=872 name="u:object_r:default_prop:s0" dev="tmpfs" ino=16580 scontext=u:r:hal_camera_default:s0 tcontext=u:object_r:default_prop:s0 tclass=file permissive=0
log1_2:05-11 17:40:17.748   872  1657 E libc    : Access denied finding property "runtime.mmitest.camera.moca"
log1_2:05-11 17:40:17.749   872  1657 I HispManager: notifyHispPreviewCamera logic cam: 0,physical cam:2
log1_2:05-11 17:40:17.752   872  1657 I HISP200Feature: sendExtendCmd: camera id 0, fbcd[1]
log1_2:05-11 17:40:17.752   872  1657 I HISP200Feature: sendExtendCmd: camera id 4, fbcd[1]
log1_2:05-11 17:40:17.752   872  1657 I Feature : setFpsParam set SUBCMD_AE_TARGET_FPS_RANGE![30, 30], intent = 3, fps = 0 Thermal = 0  m_fHwcamera=0, m_IppCtrlFlag=0
log1_2:05-11 17:40:17.752   872  1657 I HISP200Feature: setPowerSaveRawRequestStatus: flow:dual camera status:no fill raw
log1_2:05-11 17:40:17.752   872  1657 I ImagingSystem: sendCameraFovRatio: send SUBCMD_SET_CAM_FOV_RATIO ok!
log1_2:05-11 17:40:17.753   872  1657 I Processor: #Camera# [263433298]> UsecaseConfig_0
log1_2:05-11 17:40:17.768   872  1657 I Processor: #Camera# [263433298]< UsecaseConfig_0
log1_2:05-11 17:40:17.768   872  1657 I Processor: #Camera# [263448587]> UsecaseConfig_4
log1_2:05-11 17:40:17.768   872   981 I HispManager: sendCropRegionCmd: camera id 0, count:1, crop region[0 0 3648 2052]
log1_2:05-11 17:40:17.769   872  1657 I Processor: #Camera# [263448587]< UsecaseConfig_4
log1_2:05-11 17:40:17.769   857  2411 E dubaid  : [CameraHandler.cpp] SaveConfigParams# Invalid type: 4
log1_2:05-11 17:40:17.769   872  1657 I Processor: getStreamOnMsg: camera id:0.
log1_2:05-11 17:40:17.769   872  1657 I Processor: getStreamOnMsg: camera id:4.
log1_2:05-11 17:40:17.769   857  2411 E dubaid  : [CameraHandler.cpp] HandleOnConnecting# Failed to save config message parameter: -3
log1_2:05-11 17:40:17.775   872 20952 I MOCA_DEPTH: [I] dual_frequency_processor.cpp: configure : 524   read 20.000000MZH CDMA(0) cyclic error lut from file (/data/vendor/camera/MocaDepth_SensorID_23060366619212O91UED2YPS000000_20.000000MHZ_CDMA_0_SensorCrop_1_cyclic_error_lut_65536.bin) success
log1_2:05-11 17:40:17.776   872 20952 I MOCA_DEPTH: [I] dual_frequency_processor.cpp: configure : 524   read 100.000000MZH CDMA(0) cyclic error lut from file (/data/vendor/camera/MocaDepth_SensorID_23060366619212O91UED2YPS000000_100.000000MHZ_CDMA_0_SensorCrop_1_cyclic_error_lut_65536.bin) success
log1_2:05-11 17:40:17.777   872 20952 I MOCA_DEPTH: [I] dual_frequency_processor.cpp: configure : 543   read  radial to planar conversion matrix from file (/data/vendor/camera/MocaDepth_SensorID_23060366619212O91UED2YPS000000_20.000000MHZ_100.000000MHZ__CDMA_0_SensorCrop_1_144_96_3_convension_matrix_float.bin) success
log1_2:05-11 17:40:17.779   872 20952 I MOCA_DEPTH: [I] dual_frequency_processor.cpp: configure : 569   read sin and cos lookups from file (/data/vendor/camera/MocaDepth_sin_lookup_float_32768.bin, /data/vendor/camera/MocaDepth_cos_lookup_float_32768.bin) success
log1_2:05-11 17:40:17.813   854  9163 W Camera2-Parameters: set: Video size cannot be updated (from 3840 x 2160 to 1920 x 1080) when recording is active! Ignore the size update!
log1_2:05-11 17:40:17.849   872  1657 D CameraDvfs: voteIsp: HISP_CONFIG_ISP_NORMAL
log1_2:05-11 17:40:17.850   872  1657 D CameraDvfs: voteR8: HISP_CONFIG_R8_TURBO
log1_2:05-11 17:40:17.850   872  1657 I CameraPerf: limitLittleCpuDefault: CPUFREQ_OPT littleCpuFreq = 1709000
log1_2:05-11 17:40:17.851   872  1657 I ImagingSystem: #Camera# [263356660]< 3_startStreaming
log1_2:05-11 17:40:17.885   872 20959 I DeviceBaseImpl: [HWA_CAM3] getSensorModuleSpec() getSensorModuleSpec cameraID: 2
log1_2:05-11 17:40:20.698   867 20923 W CameraSource: Timed out waiting for incoming camera video frames: 0 us
[... 24 lines cut ...]
log1_2:05-11 17:40:23.003   854  1755 W Camera3-Device: notify: received notify message in error state.
log1_2:05-11 17:40:23.003   854  1755 E Camera3-Device: Camera 0: notifyError: Camera HAL reported serious device error
log1_2:05-11 17:40:23.003   854  1755 W Camera3-Device: notify: received notify message in error state.
log1_2:05-11 17:40:23.003   854  1755 E Camera3-Device: Camera 0: notifyError: Camera HAL reported serious device error
log1_2:05-11 17:40:23.003   854  1755 W Camera3-Device: notify: received notify message in error state.
log1_2:05-11 17:40:23.003   854  1755 E Camera3-Device: Camera 0: notifyError: Camera HAL reported serious device error
log1_2:05-11 17:40:23.003   854  1755 W Camera3-Device: notify: received notify message in error state.
log1_2:05-11 17:40:23.003   854  1755 E Camera3-Device: Camera 0: notifyError: Camera HAL reported serious device error
log1_2:05-11 17:40:23.003   854  1755 W Camera3-Device: notify: received notify message in error state.
log1_2:05-11 17:40:23.003   854  1755 E Camera3-Device: Camera 0: notifyError: Camera HAL reported serious device error
log1_2:05-11 17:40:23.003   854  1755 W Camera3-Device: notify: received notify message in error state.
log1_2:05-11 17:40:23.003   854  1755 E Camera3-Device: Camera 0: notifyError: Camera HAL reported serious device error
log1_2:05-11 17:40:23.047   867 20923 D CameraSource: reset: X
log1_2:05-11 17:40:23.160 20537 20684 E CAMERA_1::: stopMediaRecorder stop failed
log1_2:05-11 17:40:23.160 20537 20684 E CAMERA_1::: java.lang.RuntimeException: stop failed.
log1_2:05-11 17:40:23.160 20537 20684 E CAMERA_1:::     at android.media.MediaRecorder.native_stop(Native Method)
log1_2:05-11 17:40:23.160 20537 20684 E CAMERA_1:::     at android.media.MediaRecorder.stop(MediaRecorder.java:1886)
log1_2:05-11 17:40:23.160 20537 20684 E CAMERA_1:::     at com.google.android.cameraview.Camera1.stopMediaRecorder(Camera1.java:1543)
log1_2:05-11 17:40:23.160 20537 20684 E CAMERA_1:::     at com.google.android.cameraview.Camera1.stopRecording(Camera1.java:845)
log1_2:05-11 17:40:23.160 20537 20684 E CAMERA_1:::     at com.google.android.cameraview.Camera1.onError(Camera1.java:1597)
log1_2:05-11 17:40:23.160 20537 20684 E CAMERA_1:::     at android.media.MediaRecorder$EventHandler.handleMessage(MediaRecorder.java:1391)
log1_2:05-11 17:40:23.160 20537 20684 E CAMERA_1:::     at android.os.Handler.dispatchMessage(Handler.java:107)
log1_2:05-11 17:40:23.160 20537 20684 E CAMERA_1:::     at android.os.Looper.loop(Looper.java:213)
log1_2:05-11 17:40:23.160 20537 20684 E CAMERA_1:::     at android.os.HandlerThread.run(HandlerThread.java:67)
log1_2:05-11 17:40:23.166 20537 20684 E Camera  : Error 1

WOW-Tech, 2020-06-24:
We have the same issue, but also with the Mate Pro 20.

React-Native-Camera: 3.30
React-Native: 0.61.2
React: 16.11.0

Any hints?

WOW-Tech, 2020-06-24:
One more comment, works with front camera, but not with the normal one

WOW-Tech, 2020-06-25:
Finally we found a workaround, if you specify the quality of the video. It works ;-) So i think the detection of the max. resolution might be broken?

closed 2020-08-31 by stale[bot]

dhamzic, 2021-03-30:
> Finally we found a workaround, if you specify the quality of the video. It works ;-) So i think the detection of the max. resolution might be broken?

Works for me, cheers mate :)