react-native-camera/react-native-camera #2835: attempt to invoke virtual method 'double java.lang.double.doublevalue()' on a null object reference

anchaljain123, 2020-05-14:
# React native camera opening issue

**Ask your Question**
Getting this error when asking for record permission in device, when device has the permission,which is set after 3 to 4 times acceptance then error is gone.

**package.json**
   "react": "^16.13.0",
    "react-native": "0.61.5",
    "react-native-animatable": "^1.3.3",
    "react-native-camera": "git+https://git@github.com/react-native-community/react-native-camera.git",
    "react-native-gesture-handler": "^1.6.1",
    "react-native-image-picker": "^2.3.1",
    "react-native-reanimated": "^1.6.0"

jenskuhrjorgensen, 2020-05-14:
I get this error when tapping the camera view with newest version:
`"react-native-camera": "3.25.0"`
and
```
    react: 16.11.0 => 16.11.0
    react-native: 0.62.2 => 0.62.2
```
Tested on physical Android device: OnePlus 5T.

fabriziobertoglio1987, 2020-05-15:
https://github.com/react-native-community/react-native-camera/issues/2596

https://github.com/react-native-community/react-native-camera/issues/2047

https://github.com/react-native-community/react-native-camera/issues/1585

https://github.com/react-native-community/react-native-camera/issues/1708

I think ReadableNativeArray is used to pass data through the RN Bridge

anchaljain123, 2020-05-15:
@fabriziobertoglio1987 : What can be done to resolve the issue, should i degrade or is it breaking because permissions are setting late

fabriziobertoglio1987, 2020-05-15:
our codebase does not use that type ReadableNativeArray

https://github.com/react-native-community/react-native-camera/search?q=ReadableNativeArray&type=Code

you need to share your project implementation, please recreate this bug in a simple snippet

fabriziobertoglio1987, 2020-05-15:
https://github.com/react-native-community/react-native-camera/issues/1585#issuecomment-433194726

khattab93, 2020-06-17:
Got same error

wxb1989, 2020-06-22:
Got same error

khattab93, 2020-06-22:
@wxb1989 Check that your reference isn't null before calling its methods. For example: 
```
if (this.camera) {
      this.camera
        .recordAsync()
        .then()
        .catch()
}
```

wxb1989, 2020-06-22:
@khattab93  

I want to add a monitoring method, as follows: How should I write?
 componentDidMount() {
    if (this.rnCamera) {
      this.rnCamera
        .recordAsync()
        .then()
        .catch()
    }
    
    AppState.addEventListener('change', this.handleAppStateChange);
  }

khattab93, 2020-06-22:
@wxb1989 You need to add this event after recording a video or when?

wxb1989, 2020-06-22:
@khattab93 
 I used this.rnCamera in this method. As you said, if you don’t judge, you will get the same error as you.  
  componentDidMount() {
    AppState.addEventListener('change', this.handleAppStateChange);
  }

  componentWillUnmount() {
    AppState.removeEventListener('change', this.handleAppStateChange);
    this.rnCamera && this.rnCamera.pausePreview();
  }

  handleAppStateChange = (currentAppState) => {
    if (this.rnCamera) {
      if (currentAppState !== 'active') {
        this.rnCamera && this.rnCamera.pausePreview();
      } else {
        this.rnCamera && this.rnCamera.resumePreview();
      }
    }
  };
 <RNCamera
        ref={this.cameraRef}
        captureAudio={false}
        onBarCodeRead={this.onScanResult} />

I think it may be related to the life cycle. If I write the life cycle after the render method can I get the ref

wxb1989, 2020-06-22:
@khattab93  thank you!!!!!

closed 2020-08-28 by stale[bot]