react-native-camera/react-native-camera #2836: Missing documentation for rectOfInterest

jenskuhrjorgensen, 2020-05-14:
# Feature-Request

**Describe the Feature**
Hi
Thank you very much for the library!

I've been working on a barcode scanner, with a centered rectangle indicating to the user where he should put the barcode. I would like the scanner to ignore barcodes outside the rectangle. For this purpose I used `rectOfInterest`, and it worked out of the box and as expected on iOS. However, on Android I've been struggling to make it work. I've read [this article](https://medium.com/@jameswang1019/how-to-limit-code-scan-area-with-react-native-camera-android-and-ios-ba5e1de78155) which describes how the `rectOfInterest` (and `cameraViewDimensions`) feature has been implemented to support Android. From the article it might look like my problems are caused because I'm trying to create a scanner with a width larger than its height, violating some Android camera ratios. But I'm not sure!? I tried to `overflow: "hidden"`, but it just hides some of the camera view - then I just can't see what it is still scanning in the hidden area.

**Possible Implementations**
I think an improved documentation on this matter could save others many hours of trying to understand the differences between iOS and Android usage. Could you please add to the documentation the given limitations for use of `rectOfInterest` and `cameraViewDimensions` (if any)? And an example of its usage would be great too. 

Lastly, I'm also confused about the way the documentation is scattered: Most of the useful information about this lib I've found [here](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md), but I can see, that there is also a lot of documentation [here](https://react-native-community.github.io/react-native-camera/docs/api). I know it says "Work in progress" in some places, but it doesn't contain any information that there even exists a `rectOfInterest` or `cameraViewDimensions` prop (or the newly released version 3.25.0 with a `tap` prop), so I think it confuses more than it helps to link it in the library's README. May I ask what the the plan for/purpose of the last documentation page is?

Best regards
Jens

xelmich, 2020-05-14:
Indeed there are no clear documentation about the proper way to use rectOfInterest

anniewey, 2020-08-05:
Facing the same issue, anyone able to share an example?

jenskuhrjorgensen, 2020-10-05:
Hi bot

This is still an issue, and there seems to be quite some interest on the matter, so please don't close it.

skyrippa, 2020-10-21:
have you found any solutions on this? I'm facing the same problem, cameraViewDimensions is the total length of the screen, and the normalized values in rectOfInterest are correct, but I got no difference in the results.

jenskuhrjorgensen, 2020-10-21:
Nope, unfortunately not. :( and the documentation is still sparse.

jenskuhrjorgensen, 2021-01-06:
Bad bot!

jenskuhrjorgensen, 2021-01-13:
Thanks @alexco2

I'm not working on the project using this anymore, so not sure if your link would have helped or not. :)

fabriziobertoglio1987, 2021-03-20:
https://github.com/react-native-camera/react-native-camera/pull/2766

closed 2021-05-27 by stale[bot]