react-native-camera/react-native-camera #2837: RNCamera High Order Component

fabriziobertoglio1987, 2020-05-20:
I think we should add some sort of guide on how to create a High Order Component to wrap around [`Camera` component from `expo-camera`](https://github.com/tensorflow/tfjs/blob/5a899d358e2658325f33172a5f5df007906c9e39/tfjs-react-native/src/camera/camera_stream.tsx#L58-L73), so that developers could add their own functionalities without forking and writing a pull request

The **HOC** could use the `react-native-camera` bridge and java api. The pull request could for example add some of the `android.hardware.Camera` api to our library and expose those methods to JavaScript through the bridge. 

For example adding `android.hardware.Camera;` setter [`setDisplayOrientation()`](https://developer.android.com/reference/android/hardware/Camera#setDisplayOrientation(int)):

1. Add the Java code to expose `setDisplayOrientation()` via the JavaScript bridge to the JavaScript Developer (check my pull request https://github.com/react-native-community/react-native-camera/pull/2834 where I exposed `getSupportedPreviewFpsRange` through the bridge or https://github.com/react-native-community/react-native-camera/pull/2766 where I exposed `getCameraSettings`)
2. Create HOC like in the below example to write the javascript logic specific for your application see the [`Tensorflow` HOC](https://github.com/tensorflow/tfjs/blob/5a899d358e2658325f33172a5f5df007906c9e39/tfjs-react-native/src/camera/camera_stream.tsx#L58-L73)

```
/**
 * A higher-order-component (HOC) that augments the [Expo.Camera](https://docs.expo.io/versions/latest/sdk/camera/)
 * component with the ability to yield tensors representing the camera stream.
 *
 * Because the camera data will be consumed in the process, the original
 * camera component will not render any content. A provided by this component
 * is used to render the camera preview.
 *
 * Notably the component allows on-the-fly resizing of the camera image to
 * smaller dimensions, this speeds up data transfer between the native and
 * javascript threads immensely.
 *
 * __In addition to__ all the props taken by Expo.Camera. The returned
 * component takes the following props
 *
```

I would like to know what @renatobenks, @cristianoccazinsp  and @sibelius thinks about implementing such a HOC to add functionalities to RNCamera :smiley: 

The only issue with using the HOC is that the data stream in JavaScript may cause some performance issues, but as long as you are setting an instance variable, changing property and not actually postprocesssing the video in JavaScript, then you will not need native functionality ...

Thanks for reading
Fabrizio

sibelius, 2020-05-20:
What do you think of using hooks instead of HOCs?

like this one https://github.com/reime005/react-native-camera-hooks

each hook could be "bind" to a native capability

cristianoccazinsp, 2020-05-20:
Just to be clear, you are suggesting that the library expose some more native methods so it is easier to extend on the JS side. This makes sense, although special care needs to be taken since iOS and Android APIs are quite different!

About actual HOC components, I think making those should be straight forward to anyone with React knowledge? 

For hooks, I haven't really got used to them yet, but I know they are the "cool" alternative to HOCs. However, mixing hooks and class based components is difficult, so you may need to write one guide for each type of implementation.

sibelius, 2020-05-20:
we need to move to functional components + hooks

and avoid class based on

renatobenks, 2020-05-20:
It does sound making sense @fabriziobertoglio1987 what you're describing here, but I'm not sure that we should be thinking on to build that by using the HOC approach instead of the hooks API as @sibelius told above.

So, in fact, I can't see any further advantage to make it to work besides hooks API, because it seems that would be enough and a good fit as well for what you're expecting to use from the camera, I guess at least. But just to make, which kinds of features @fabriziobertoglio1987 do you think that HOCs would be capable to expose over the hooks API?

fabriziobertoglio1987, 2020-05-20:
Thanks @cristianoccazinsp 
>Just to be clear, you are suggesting that the library expose some more native methods so it is easier to extend on the JS side. 

Yes. 

>This makes sense, although special care needs to be taken since iOS and Android APIs are quite different!

We could offer two options:

1) The `RNCamera` component for basic usage
2) A separate opensource library which would expose a `High Order Component` built around `RNCamera`. The repository would include explanation on how to add native methods (for ex. `setCameraZoom`). The developers would just add methods to the Java/ObjectiveC/RNBridge Application Programming Interface and open a pull request to merge those additions to `react-native-camera`. The additions would not trigger regressions as they would just add one more endpoint to the API.

>For hooks, I haven't really got used to them yet, but I know they are the "cool" alternative to HOCs. However, mixing hooks and class based components is difficult, so you may need to write one guide for each type of implementation.

I am not very skilled in Hooks and I am new to functional programming.

My basic understanding of `High Order Functions` is that they take one or more functions as arguments and returns a function as its result.

Thanks @sibelius 
>and avoid class based components

My understanding is that `facebook/react` decided to avoid Inheritance/Mixins. 
As inheritance and mixins are fundamental for Object Oriented Programming, class based components have limited use.

The alternative to Inheritance/Mixins was using `High Order Component`, but later react team developed and incentivated the adoption of react `hooks`.

>What do you think of using hooks instead of HOCs? like this one https://github.com/reime005/react-native-camera-hooks
>each hook could be "bind" to a native capability

The main purpose of the HOCs would be teaching developers to enhance the `react-native-camera` libary without making a pull request.

They can write their own Javascript Logic in `react-native-camera` without changing the library code. They can add Javascript logic to the `react-native-camera` re-using the existing Android/iOS APIs from `react-native-camera` or add the missing APIs.

Hooks would require us to either add a new javascript layer on top of our library or rewrite the existing JavaScript logic. The first option would introduce more complexity and issues, the second one would still require a considerable amount of work.

HOC requires no work ([sourcecode](https://github.com/tensorflow/tfjs/blob/5a899d358e2658325f33172a5f5df007906c9e39/tfjs-react-native/src/camera/camera_stream.tsx#L58-L73))

No developers will invest the time creating HOC based on `RNCamera` if their pull request are merged in the main repository, but only after agreement to follow [`lean core`](https://github.com/facebook/react-native/issues/23313) which would mean not merging non essential pull requests.

I'll still contribute to the library in different ways in all scenarios. Thanks a lot.

fabriziobertoglio1987, 2020-05-21:
@cristianoccazinsp 

I believe it is not worth it to inherit or use a high order component to patch the `react-native-camera` library. It is just too complex and you would inherit all the issues from `react-native-camera`.

The best choice for a react-native Camera application is using a Camera component for relatively simple use cases and your own Camera component for more complex use case

https://reactnative.dev/docs/native-components-android

For example, you can use `react-native-camera` for iOS and your own android component for android. 

I will be easier to support functionalities like pinch to zoom and many others natively as you wrote the native library... Technically you could fork react-native-camera and delete not essential functionalities or fork react-native-camera-kit

closed 2020-05-31 by fabriziobertoglio1987