react-native-camera/react-native-camera #2839: Unable to render RNCamera component

malcolmmcswain, 2020-05-21:
I have the following component:
```
import { RNCamera } from 'react-native-camera';

const Home: React.FC<HomeProps> = ({ navigation }) => {
  const cameraRef = useRef();

  return (
    <Container>
      <Content style={styles.content}>
        <RNCamera
          ref={cameraRef}
          style={{flex: 1, width: '100%', height: '100%'}}
          type={RNCamera.Constants.Type.back}
          captureAudio={false}
        />
      </Content>
    </Container>
  );
};
```
On iOS, I get no warnings from React Native, I am prompted for camera permissions, but I cannot see the camera being rendered at all, can't even find it in the inspector or find any indications in the log that there was a failed attempt at rendering the camera. I only added the NSCameraUsageDescription in my Info.plist, since I just plan to use the camera for scanning QR codes. Any suggestions here? I've tried nearly every solution I could find online.

jaworek, 2020-05-26:
I'm having exact same problem. I created empty app using `npx react-native init`.
I followed instructions on how to automatically install RNCamera.

System: MacOS 10.15.4
Node: 14.3.0
react: 16.11.0
react-native: 0.62.2
react-native-camera: 3.26.0

Installed on iPhone SE (2016) with iOS 13.4.1.

jaworek, 2020-05-26:
@malcolmmcswain try this code:

```
import React from 'react';
import {View} from 'react-native';

import {RNCamera} from 'react-native-camera';

const App = () => {
  return (
    <View style={{flex: 1}}>
      <RNCamera
        style={[{flex: 1, width: '100%', overflow: 'hidden'}]}
        type={RNCamera.Constants.Type.back}
        flashMode={RNCamera.Constants.FlashMode.on}
        captureAudio={false}
        androidCameraPermissionOptions={{
          title: 'Permission to use camera',
          message: 'We need your permission to use your camera',
          buttonPositive: 'Ok',
          buttonNegative: 'Cancel',
        }}
      />
    </View>
  );
};

export default App;
```

I noticed that if RNCamera is nested in multiple ```View``` components, every one of them needs to have applied style ```flex: 1``` and ```width: "100%"```. I don't know exactly why. If I figure out a better way to set it up, I'll post an update.

malcolmmcswain, 2020-05-28:
Thank you so much for your comment. You helped me fix an issue I've been struggling with for nearly a month. I'm now able to render the camera, however, when I set the height to a fixed integer, it doesn't affect the actual height of the component. Any idea why this is?

malcolmmcswain, 2020-05-28:
[image]

malcolmmcswain, 2020-05-28:
Update: I set the flex of the actual RNCamera component to 0 and then the height was applied. Closing this issue now. There should be something in the docs about this though because it was a really frustrating experience to follow the installation instructions word for word and then just get a blank view.

closed 2020-05-28 by malcolmmcswain

jaworek, 2020-05-28:
I'm glad that I could help! 😄 
You can get rid of ```flex: 0``` from the RNCamera component entirely. From what I understand flex property takes precedence over height or width.

You probably found this section in documentation:
https://reactnative.dev/docs/height-and-width
[image]

It does not explicitly state it, but I think this is what they are suggesting here. Probably additional warning would solve this problem for many.