react-native-camera/react-native-camera #2843: Can't get it to work, giving Error: 'Could not resolve project :react-native-camera'

soberservicesguy, 2020-05-26:
# Bug Report

**To Do First**

- [Yes ] Did you try latest release?
- [ Yes] Did you try master?
- [ Yes] Did you look for existing matching issues?

**Platforms**
Ubuntu

 Android

**Versions**

- Android:
- iOS:
- react-native-camera:
- react-native:4.8.0
- react:
  

**Description/Current Behaviour**
I did not install properly since it fails the build after that

**Expected Behaviour**

**Steps to Reproduce**
I am trying to install React native camera in Ubuntu for android and I ran the following to install 

npm install react-native-camera --save

then I ran `npx react-native run-android` which gave the following error
```$ npx react-native run-android
info Running jetifier to migrate libraries to AndroidX. You can disable it using "--no-jetifier" flag.
Jetifier found 1343 file(s) to forward-jetify. Using 4 workers...
info Starting JS server...
info Installing the app...

Deprecated Gradle features were used in this build, making it incompatible with Gradle 7.0.
Use '--warning-mode all' to show the individual deprecation warnings.
See https://docs.gradle.org/6.0.1/userguide/command_line_interface.html#sec:command_line_warnings

FAILURE: Build failed with an exception.

* What went wrong:
Could not determine the dependencies of task ':app:compileDebugJavaWithJavac'.
> Could not resolve all task dependencies for configuration ':app:debugCompileClasspath'.
   > Could not resolve project :react-native-camera.
     Required by:
         project :app
      > Cannot choose between the following variants of project :react-native-camera:
          - generalDebugApiElements
          - mlkitDebugApiElements
        All of them match the consumer attributes:
          - Variant 'generalDebugApiElements' capability Guacamole_UI_kit:react-native-camera:unspecified:
              - Unmatched attributes:
                  - Found com.android.build.api.attributes.VariantAttr 'generalDebug' but wasn't required.
                  - Found react-native-camera 'general' but wasn't required.
              - Compatible attributes:
                  - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found compatible value 'debug'.
                  - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found compatible value 'Aar'.
                  - Required org.gradle.usage 'java-api' and found compatible value 'java-api'.
          - Variant 'mlkitDebugApiElements' capability Guacamole_UI_kit:react-native-camera:unspecified:
              - Unmatched attributes:
                  - Found com.android.build.api.attributes.VariantAttr 'mlkitDebug' but wasn't required.
                  - Found react-native-camera 'mlkit' but wasn't required.
              - Compatible attributes:
                  - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found compatible value 'debug'.
                  - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found compatible value 'Aar'.
[... 91 lines cut ...]
Could not determine the dependencies of task ':react-native-camera:compileGeneralDebugAidl'.
> Could not resolve all task dependencies for configuration ':react-native-camera:generalDebugCompileClasspath'.
   > Could not find com.google.android.gms:play-services-vision:16.1.0.
     Required by:
         project :react-native-camera

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 17s

    at makeError (/home/arsalan/Work_stuff/Frontend_apps/React_native_apps/Portfolio_apps/Guacamole_UI_kit/node_modules/execa/index.js:174:9)
    at /home/arsalan/Work_stuff/Frontend_apps/React_native_apps/Portfolio_apps/Guacamole_UI_kit/node_modules/execa/index.js:278:16
    at processTicksAndRejections (internal/process/task_queues.js:97:5)
    at async runOnAllDevices (/home/arsalan/Work_stuff/Frontend_apps/React_native_apps/Portfolio_apps/Guacamole_UI_kit/node_modules/@react-native-community/cli-platform-android/build/commands/runAndroid/runOnAllDevices.js:94:5)
    at async Command.handleAction (/home/arsalan/Work_stuff/Frontend_apps/React_native_apps/Portfolio_apps/Guacamole_UI_kit/node_modules/react-native/node_modules/@react-native-community/cli/build/index.js:186:9)
```

**Additionals**
If you can help me out to get it working, that will be huge favor. Thanks a lot.

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

trimitrtechnologies, 2020-05-26:
in android/app/build.gradle

Add below line inside defaultConfige

`defaultConfig {`
`        ...`
` missingDimensionStrategy 'react-native-camera', 'general' // <--- insert this line`
`    }`

wi100sh, 2020-06-03:
Thanks a lot, this solve my problem.

menufour, 2020-06-13:
Why this not worked for me?

ammarRajabA, 2020-09-03:
It would be a good idea for this to be in documentation

sibelius, 2020-09-03:
feel free to send a PR

Olovorr, 2020-10-21:
@sibelius this is already in the documentation :) 
[image]

closed 2021-01-13 by stale[bot]

RavikumarRapolu, 2021-05-04:
Iam getting below errors after installing React native camera 

Error: Command failed: gradlew.bat app:installDebug -PreactNativeDevServerPort=8081
[Fatal Error] metadata-extractor-2.11.0.pom:2:1: Content is not allowed in prolog.
[Fatal Error] legacy-support-v4-1.0.0.pom:2:1: Content is not allowed in prolog.
[Fatal Error] legacy-support-v4-1.0.0.pom:2:1: Content is not allowed in prolog.
[Fatal Error] core-3.3.3.pom:2:1: Content is not allowed in prolog.
[Fatal Error] exifinterface-1.3.2.pom:2:1: Content is not allowed in prolog.
[Fatal Error] exifinterface-1.3.2.pom:2:1: Content is not allowed in prolog.

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:checkDebugAarMetadata'.
> Could not resolve all files for configuration ':app:debugRuntimeClasspath'.
   > Could not resolve androidx.legacy:legacy-support-v4:1.0.0.
     Required by:
         project :app
         project :app > project :react-native-camera
      > Could not resolve androidx.legacy:legacy-support-v4:1.0.0.
         > Could not parse POM https://dl.google.com/dl/android/maven2/androidx/legacy/legacy-support-v4/1.0.0/legacy-support-v4-1.0.0.pom
            > Content is not allowed in prolog.
      > Could not resolve androidx.legacy:legacy-support-v4:1.0.0.
         > Could not parse POM https://maven.google.com/androidx/legacy/legacy-support-v4/1.0.0/legacy-support-v4-1.0.0.pom
            > Content is not allowed in prolog.
   > Could not resolve com.drewnoakes:metadata-extractor:2.11.0.
     Required by:
         project :app
         project :app > project :react-native-camera
      > Could not resolve com.drewnoakes:metadata-extractor:2.11.0.
         > Could not parse POM https://jcenter.bintray.com/com/drewnoakes/metadata-extractor/2.11.0/metadata-extractor-2.11.0.pom
            > Content is not allowed in prolog.
   > Could not resolve com.google.zxing:core:3.3.3.
     Required by:
         project :app > project :react-native-camera
      > Could not resolve com.google.zxing:core:3.3.3.
         > Could not parse POM https://jcenter.bintray.com/com/google/zxing/core/3.3.3/core-3.3.3.pom
            > Content is not allowed in prolog.
   > Could not resolve androidx.exifinterface:exifinterface:1.3.2.
     Required by:
         project :app > project :react-native-camera
      > Could not resolve androidx.exifinterface:exifinterface:1.3.2.
         > Could not parse POM https://dl.google.com/dl/android/maven2/androidx/exifinterface/exifinterface/1.3.2/exifinterface-1.3.2.pom
            > Content is not allowed in prolog.
      > Could not resolve androidx.exifinterface:exifinterface:1.3.2.
         > Could not parse POM https://maven.google.com/androidx/exifinterface/exifinterface/1.3.2/exifinterface-1.3.2.pom
            > Content is not allowed in prolog.