react-native-camera/react-native-camera #2844: Why only dispatch events every 500ms in iOS ?

lumixraku, 2020-05-27:
# Question

**Ask your Question**
Why only dispatch events every 500ms in iOS ? 

I thought iOS would have much better experience than Android. But actually it has a freq limit. 

I changed this time gap to 50ms, And face detect seems works fine.

I just do not understand why iOS would have this limit.

sibelius, 2020-05-27:
where is the limit set?
I can we can make this lower or configurable

lumixraku, 2020-05-28:
> where is the limit set?
> I can we can make this lower or configurable

In RNCamera.m       Line 2236

tomsotte, 2020-05-28:
It's one of the thing I wanted to tackle on this PR #2838, because it's very useful for real-time face contours detection.

sibelius, 2020-05-28:
I think we can change to a new better default

closed 2020-05-29 by lumixraku