react-native-camera/react-native-camera #2845: disable ML Kit face-detection

MujtabaFR, 2020-05-28:
Hello,
I'm using  **react-native-camera** for barcode scanning in my app, not using any face detection

the issue is when using **ML Kit**.. the final **apk size increases by 10Mb** for each architecture.

The cause is `face detection models` and native library `libface_detector_v2_jni` that is included when enabling **ML Kit**

So is there any way to use ML Kit but exclude face detection ?

this image shows the deference in apk size

[image]

Dkokkonas, 2020-06-12:
Hey I am facing the same issues . Did you find a way to resolve this?

Return-1, 2020-06-12:
Im facing a similar issue, not using ML Kit at all, would like to disable alltogether as im only using `react-native-camera` to allow for `react-native-qrcode-scanner` to work. This is a lot of bloat for camera-only usage : /

MujtabaFR, 2020-06-12:
@Dkokkonas I didn't find any solution ..
So I had to use react-native-camera without **MLKit** all together

sibelius, 2020-06-12:
I think you can change the `missingDimensionStrategy` on your android config to avoid having mlkit

https://react-native-community.github.io/react-native-camera/docs/installation#modifying-buildgradle

MujtabaFR, 2020-06-15:
@sibelius  I think it's easy and clear how to enable/disable MLKit
but is there any way to enable (MLKit for barcode) and disable (MLKit for face detection) ?

as the extra size is caused by (face detection) only

sibelius, 2020-06-15:
@MujtabaFR can you investigate if this is possible in the native side?

so we can document how to strip the face detection part?

AnagramEngineering, 2020-09-28:
Add this to your build.gradle 

    implementation (project(':react-native-camera')){
        exclude group: 'com.google.firebase', module: 'firebase-ml-vision-face-model'
    }
I can confirm both, qr scanning working and reduction of apk size.